chipsalliance/riscv-dv #714: [Pygen]: import issue while integrating rv32imc core setting file to the overall run flow

aneels3, 2020-09-18:
The current python generator supports only RV32I instructions and I am using the core setting file for rv32i like this
[image]

The alias **rcs** is used throughout the pygen_src source files.
In order to make the generator generic with other extensions using the target switch of run.py, the import should be done in such a way that the core setting import should match with the target switch (rv32i or rv32imc).

[image]

While doing the above approach I have encountered a cyclic dependency with the **riscv_instr_pkg** class as I am importing the core setting file in pkg class as well.

Need some help to make this generic with the target switch, or a possible workaround for this issue.

hodjat91, 2020-09-18:
Hi @aneels3 

Thanks for opening this issue. Now I kind of understand where the problem comes from. To be able to fix it, and to fully understand it, could you please share the code and the specific error with me? I assume you haven't yet created a PR for imc extension, so I think I don't have access to the code that generates the circular dependency. 

Cheers,
Hodjat

aneels3, 2020-09-18:
Hi @hodjat91 
You can find the current code in the **[pygen_dev](https://github.com/saurabhsingh-pvips/riscv-dv/tree/pygen_dev/pygen/pygen_src)** branch.

Thanks!

hodjat91, 2020-09-19:
Hi @aneels3 

I was able to solve the circular dependency. In general, fixing the circular import hole in Python is a bit tricky. This specific circular import should get resolved as follows:
1- In file `pygen/pygen_src/test/riscv_instr_base_test.py`, import the `riscv_instr_pkg` before importing cfg (right after `sys.path.append("pygen")`: `from pygen_src.riscv_instr_pkg import *`. This way, we import `riscv_instr_pkg` before `cfg`.
2- In `pygen/pygen_src/riscv_instr_pkg.py`, move these import statements from top of the file into the beginning of `class riscv_instr_pkg` (please note that the `global rcs` needs to be added): 
```
global rcs
from pygen_src.riscv_instr_gen_config import cfg
if cfg.argv.target == "rv32i":
    from pygen_src.target.rv32i import riscv_core_setting as rcs
if cfg.argv.target == "rv32imc":
    from pygen_src.target.rv32imc import riscv_core_setting as rcs
def __init__(self):
```

This will solve the circular dependency: By the time that `pygen/pygen_src/riscv_instr_gen_config.py` wants to import stuff from `riscv_instr_pkg`, things are already defined.

I tested this approach and it works fine. Just one small issue with the modified `run.py` by you: a space is missing after `--target=%s` in `do_simulate` method. Also the `base_testlist.yaml` file in your `pygen_dev` branch has some unknown arguments that needs to be fixed (for example, `isa` is not recognized by the `run.py`). 

Please let me know if you have any questions, or if the approach isn't clear/doesn't make sense to you. 

-Hodjat

aneels3, 2020-09-21:
Hi @hodjat91 
Thanks for this. It totally makes sense to me now.
I am closing this issue for now.

saurabhsingh-pvips, 2020-09-28:
@hodjat91 

Getting cfg related error while generating coverage for pyflow. The error looks like a consequence of changes made to resolve this issue.
```
from pygen_src.isa.riscv_cov_instr import riscv_cov_instr
  File "pygen/pygen_src/isa/riscv_cov_instr.py", line 24, in <module>
    if cfg.argv.target == "rv32i":
NameError: name 'cfg' is not defined
```
Run Command: `python3 cov.py --simulator pyflow --enable_visualization --dir out_2020-09-28/ovpsim_sim/`

Can you take a look? If this issue is not related with this thread, let me know, I'll file a separate issue for the same.

hodjat91, 2020-09-28:
Hi @saurabhsingh-pvips 

Are you seeing this with the most current master branch? Or I should run your pygen_dev branch to reproduce this error?

saurabhsingh-pvips, 2020-09-29:
Hi @hodjat91 

This is observed in pygen_dev branch only. Working fine in master. You can checkout [pygen_dev ](https://github.com/saurabhsingh-pvips/riscv-dv/tree/pygen_dev)to reproduce error.

hodjat91, 2020-09-29:
Hi @saurabhsingh-pvips 

I ran the pygen_dev branch (I couldn't find your out_2020-09-28/ovpsim_sim/ directory in the dev_branch, so I did one myself with Spike). I don't get such error actually. It just throws error regarding some "unrecognized arguments" which means that the part of code that you pointed out was successfully executed. Could you please make sure that current head of pygen_dev actually generates such error? Also, after confirming that pygen_dev branch generates this error, please make sure to include the out_2020-09-28/ovpsim_sim/ directory as well in your branch so we can exactly run the same thing (however, that shouldn't be the problem). 

Thanks,
Hodjat

saurabhsingh-pvips, 2020-09-29:
Hi @hodjat91 ,

I double checked for the head, it's pointing to the latest commit on pygen_dev and I am still getting the error. Created a new branch, coverage_cfg_issue on top of pygen_dev and added the required out dir and coverage.log for error log. Please checkout [coverage_cfg_issue](https://github.com/saurabhsingh-pvips/riscv-dv/tree/coverage_cfg_issue). 

Let me know if I could assist with something else.

hodjat91, 2020-09-29:
Hi @saurabhsingh-pvips 

Thanks for confirming the issue and creating the new branch. I looked at the problem, here are the required steps to fix this:
1- In general, the issue that you reported isn't a "circular dependency" issue: as you can see in the output, the error complains about a NameError: `NameError: name 'cfg' is not defined`, and that is because now you import different `rcs` based on the `cfg.target`, but `cfg` itself has not been imported yet. So you need to simply import it before using: `from pygen.pygen_src.riscv_instr_gen_config import cfg`.

2- However, after fixing this, there are two new issues that come up in parallel: 
2.1- Since in your implementation, you've involved `cfg` in `riscv_cov_instr.py` to get the correct `rcs`, and because all of the command line argument are parsed in the `riscv_instr_gen_config.py` file, we need to move the command line arguments of `riscv_instr_cov_test.py` to the `riscv_instr_gen_config.py` (similar to what we did for `riscv_instr_base_test.py`). Please note that without this, we will get unrecognized argument error as arguments have been parsed already in the `riscv_instr_gen_config.py` file.  
2.2- A circular dependency issue come up that will be fixed by applying this simple change: In `riscv_instr_cov_test.py`, you need to move the `from pygen_src.riscv_instr_pkg import *` to right after the `sys.path.append("pygen/")` to avoid such circular dependency. 

After doing these changes, it works fine for me. I will create a PR for this on your repo. Please let me know if it doesn't make sense of more clarification is needed. 

Cheers,
Hodjat

aneels3, 2020-09-30:
Hi @hodjat91 
I totally get the approach you suggested and I have made the changes according to it [here](https://github.com/saurabhsingh-pvips/riscv-dv/commit/ef7eca3c9964e518993a0a89f7fb3dac82dea4b4)
Now it's working fine for me. 
Can you confirm these changes from your end?
If it looks good to you then I will merge these changes in the open PR #724 

Thanks for the help!

Regards,
~Anil

hodjat91, 2020-09-30:
Hi @aneels3 

My pleasure, It LGTM. Please go ahead and proceed with merging these changes in the open PR #724 

Cheers,
Hodjat

closed 2020-09-30 by hodjat91