chipsalliance/riscv-dv #718: Single-double precision is not distinguished at the coverpoint FP_SPECIAL_VALUE_CP

blackcatruri, 2020-09-22:
Although in file riscv_instr_cover_group.sv, the definition of FP_SPECIAL_VALUE_CP distinguishes between single and double precision cases, both types of coverpoints exist in the covergroup of floating-point instruction when making coverage statistics.

weicaiyang, 2020-09-24:
> Although in file riscv_instr_cover_group.sv, the definition of FP_SPECIAL_VALUE_CP distinguishes between single and double precision cases, both types of coverpoints exist in the covergroup of floating-point instruction when making coverage statistics.

@blackcatruri 
yes, it's shown in coverage statistics, but since the weight is 0, it doesn't contribute to overall coverage. In below example, fd is unsigned, `cp_fd_sign` has 0 weight, so it doesn't affect the overall coverage even this coverpoint is not 100%.
[image]

blackcatruri, 2020-09-24:
ok，i see，thanks

closed 2020-09-24 by weicaiyang