chipsalliance/riscv-dv #721: UVM Error - coverage test fails for "ml" target because of the inserted illegal instruction stream

dariuszst, 2020-09-23:
The SV simulation of the coverage test ends with a list of UVM errors:
` KERNEL: UVM_ERROR riscv-dv/test/riscv_instr_cov_test.sv(84) @ 0: uvm_test_top [uvm_test_top] Found unexpected illegal instr: unknown [000000008000bb5c,unknown,,,44e08973,,unknown,,`

This happens because the testlist.yaml for the "ml" target sets the "+illegal_instr_ratio=5" parameter that causes in same cases inserting of the illegal instruction stream during assembly generation:

```
# KERNEL: UVM_INFO riscv_asm_program_gen.sv(116) @ 0: reporter [asm_gen] Generating main program instruction stream...done
# KERNEL: UVM_INFO riscv-dv/src/riscv_instr_sequence.sv(327) @ 0: reporter@@sub_3 [sub_3] Injecting 0 illegal instructions, ratio 5/100
# KERNEL: UVM_INFO riscv-dv/src/riscv_instr_sequence.sv(340) @ 0: reporter@@sub_3 [sub_3] Injecting 0 HINT instructions, ratio 5/100
# KERNEL: UVM_INFO riscv-dv/src/riscv_instr_sequence.sv(327) @ 0: reporter@@sub_1 [sub_1] Injecting 31 illegal instructions, ratio 5/100
# KERNEL: UVM_INFO riscv-dv/src/riscv_instr_sequence.sv(340) @ 0: reporter@@sub_1 [sub_1] Injecting 31 HINT instructions, ratio 5/100
```

The generated assembly file risc_rand_test_0.S contains for exemple lines:
```    
 .4byte 0x44e08973 # kIllegalSystemInstr
```
This is properly simulated by spike and written to the risc_rand_test_0.log file:
```
000000008000bb5c,unknown,,,44e08973,,unknown,,
```
The problem is that the name of the log doesn't contain "illegal" string - that is used to set the expect_illegal_instr variable in:
https://github.com/google/riscv-dv/blob/8a6a399ed5268130f30879d5548d97b477f21bdf/test/riscv_instr_cov_test.sv#L45-L47

and finally the UVM error is generated:
https://github.com/google/riscv-dv/blob/8a6a399ed5268130f30879d5548d97b477f21bdf/test/riscv_instr_cov_test.sv#L82-L84

If the illegal instructions stream is expected in this case the name of the generated program should be changed or alternatively other method of marking that illegal instructions are allowed should be implemented.

udinator, 2020-09-28:
Thanks for bringing up this issue @dariuszst. I think you are right, probably should not have functional coverage for illegal instructions depend on the name of the test file.
I'll work on a fix for this.

closed 2020-10-12 by taoliug

dariuszst, 2020-10-15:
Unfortunately, the fix #729 doesn't solve this problem. Now, the coverage test will accept illegal instructions if the spike log file name contains "illegal" or "unknown" string:

` if (uvm_is_match("*illegal*", trace_csv[i]) || uvm_is_match("*unknown*", trace_csv[i]) ) begin...`

Hovewer, as defined in the testlist.yam file, the run.py will generate the log file named:  risc_rand_test_0.log so coverage test will also fail. Am I right?

PR #729 (2020-10-12): Fix coverage issue for ml target
Fix #721