chipsalliance/riscv-dv #722: SPIKE timeout error: trap_store_address_misaligned or trap_store_fault_address error.

saurabhsingh-pvips, 2020-09-25:
While simulating RV32I assembly file generated using pyflow, getting timeout error due to following two errors:

- core   0: exception trap_store_address_misaligned, epc 0xffffffff80000318
- core   0: exception trap_store_access_fault, epc 0xffffffff80000320

A closer look suggests this error happens when mtvec_handler block of the asm program file starts executing. No such issue is encountered while simulating with OVPsim.

Steps to reproduce: 
- Checkout this branch : [pygen_dev](https://github.com/saurabhsingh-pvips/riscv-dv/tree/pygen_dev)
- Run command: `python3 run.py --test riscv_arithmetic_basic_test --target rv32i --iterations 1 --simulator pyflow`

Tried the following things at my end by observing the comments in #250 ,
- debug_mode disabled via core_setting. (mentioned [here](https://github.com/google/riscv-dv/issues/250#issuecomment-545263288))
- Switches --isa rv32i --mabi ilp32 (mentioned [here](https://github.com/google/riscv-dv/issues/250#issuecomment-545561499))

These didn't help.

Maybe we are missing something while generating the asm_file or something is generated incorrectly.
This issue is also observed for target RV32IMC. (Use switch `--target rv32imc`).

Attaching assembly file & spike_sim log for both, store_fault & store_misaligned errors.

[store_misaligned_asm.log](https://github.com/google/riscv-dv/files/5282910/store_misaligned_asm.log)
[store_misaligned_spike_sim.log](https://github.com/google/riscv-dv/files/5282919/store_misaligned_spike_sim.log)

[store_fault_asm.log](https://github.com/google/riscv-dv/files/5282911/store_fault_asm.log)
[store_fault_spike_sim.log](https://github.com/google/riscv-dv/files/5282921/store_fault_spike_sim.log)

udinator, 2020-09-28:
Hi @saurabhsingh-pvips thanks for raising this issue, I will take a look into this.
Can you please paste the OVPsim logs as well?

saurabhsingh-pvips, 2020-09-29:
@udinator 
Some server error at my end, couldn't access the previous logs & files. Hence uploading a new set of files:

asm_file: [riscv_arithmetic_basic_test_0.log](https://github.com/google/riscv-dv/files/5297220/riscv_arithmetic_basic_test_0.log) 
spike_sim_log(misaligned error):  [SPIKE_riscv_arithmetic_basic_test.0.log](https://github.com/google/riscv-dv/files/5297222/SPIKE_riscv_arithmetic_basic_test.0.log)
ovpsim_log: [OVPSIM_riscv_arithmetic_basic_test.0.log](https://github.com/google/riscv-dv/files/5297224/OVPSIM_riscv_arithmetic_basic_test.0.log)
Disassembled code: [DSM_riscv_arithmetic_basic_test_0.log](https://github.com/google/riscv-dv/files/5297229/DSM_riscv_arithmetic_basic_test_0.log)

udinator, 2020-09-30:
Thanks for the updated files!

There are two issues here causing this Spike error, I'll explain both:

1) In `riscv_asm_program_gen::init_gpr(self)`, both of the `cfg.sp` and `cfg.tp` registers need to be excluded from randomization. What is happening in your error case is that in the generated `<kernel_sp>` code section, the random `tp` register is being  initialized to the address of `<kernel_stack_end>` section, but then in GPR initialization, it (as well as `sp`) are being overridden with some random initial values, thus immediately writing the incorrect value into these two registers. You can take a look at this function in the SV-side for reference. This point should be able to be fixed pretty easily.

2) The second issue might be a bit more difficult to overcome. In the example files that you just recently sent me, it looks like `t3` is being randomly chosen to act as the `tp` register, so I'll use this as example. Throughout the `<main>` section of code, you can see various instructions that use `t3` as the destination register, thus overwriting the value stored in there over and over again. 
The same is happening with all the other registers in the `cfg.reserved_regs` list, they are all being used as destination register even though this should not happen.  This constraint is dealt with in the `riscv_instr_stream::randomize_gpr()` function on the SV-side, you can see that there is a `foreach` loop constraint that prevents any destination register from being set to one of the globally reserved registers in `cfg.reserved_regs`. 
However on the Python side in `riscv_instr_stream::randomize_gpr()` function, we do not have any of these necessary inline random constraints due the lack of support in PyVSC at the time of development.
As a result, to fix this bug we will need the inline randomization support from PyVSC, workaround for this might be tricky. Adding @hodjat91 for any insights.

Please let me know if anything above does not make sense.

hodjat91, 2020-09-30:
Hi @udinator and @saurabhsingh-pvips 

Regarding the second issue: good thing is that inline randomization is actually supported in PyVSC (however, in the current version it seems like it doesn't support a list of "enum_type" variables for inline randomization). While adding the support for enum_type should be straightforward, I believe as a possible workaround for now, we can implement the same `randomize_gpr()` along with the `foreach` loop constraints on a list of register "names" (instead of a list of register enums). Later on, once we get the randomized "names", we can convert them to their enum_type equivalent. We can also use `@vsc.dynamic_constraint` to define the additional constraints on `cfg.reserved_regs` (for example, making sure that destination register can't be any of the `cfg.reserved_regs`).

Does it make sense?

saurabhsingh-pvips, 2020-09-30:
Hi @udinator ,
Thanks.
Your comments make the problem clear. We understand what's going wrong, I and @aneels3 are working on it at our end. 
@hodjat91 thanks for the workaround tip, we'll try and get this fixed. 
It might take a while. We'll keep you posted.

hodjat91, 2020-09-30:
Hi @saurabhsingh-pvips 

No problem, please let me know if you face any issue with the workaround. 

Hodjat

saurabhsingh-pvips, 2020-10-06:
Fix added [here](https://github.com/saurabhsingh-pvips/riscv-dv/commit/2684880d1b4bfef98f6fdf3a666578943ec19c46). 
Closing this issue now.

closed 2020-10-06 by saurabhsingh-pvips

moyouth, 2023-10-17:
> @udinator Some server error at my end, couldn't access the previous logs & files. Hence uploading a new set of files:
>
> asm_file: [...]

"I want to know how these detailed instructions in the OVPIsm logs are displayed through the configuration file."
[image]

moyouth, 2023-10-18:
> @udinator Some server error at my end, couldn't access the previous logs & files. Hence uploading a new set of files:
>
> asm_file: [...]

Run command: python3 run.py --test riscv_arithmetic_basic_test --target rv64gc --iterations 1 --simulator vcs --iss ovpsim

riscv_arithmetic_basic_test_0.log

[image]

iss.yaml

[image]
why ovpsim does not support "--trace --tracechange"