chipsalliance/riscv-dv #725: Wrong assembly code is generated in some cases by SV generator that causes unexpected spike simulation error.

dariuszst, 2020-09-29:
The randomization of cfg.scratch_reg register seems to be constrained only by following constraint:
https://github.com/google/riscv-dv/blob/8a6a399ed5268130f30879d5548d97b477f21bdf/src/riscv_instr_gen_config.sv#L414-L418

This causes that scratch_reg can be in some cases randomized to be the GP register that is the same as x3:
https://github.com/google/riscv-dv/blob/8a6a399ed5268130f30879d5548d97b477f21bdf/src/riscv_asm_program_gen.sv#L1151

During the exit procedure the GP register is assigned with 1 but then it is overridden by assigning "write_tohost" subprogram address to the x3 register.

```
test_done:        
                  li gp, 1
                  ecall
...
ecall_handler:    
                  la x25, _start
                  sw x0, 0(x25)
                  sw x1, 4(x25)
                 ...
                  sw x31, 124(x25)
                  la x3, write_tohost
                  jalr x0, x3, 0

...
write_tohost:     
                  sw gp, tohost, t5

_exit:            
                  j write_tohost
```

This causes that spike ends simulation with unexpected error.  See https://github.com/riscv/riscv-isa-sim/issues/555
The complete assembly program and simulator log is attached in a zip file:
[rv32i_riscv_jump_stress_test.zip](https://github.com/google/riscv-dv/files/5298312/rv32i_riscv_jump_stress_test.zip)

udinator, 2020-09-29:
Thanks for opening the issue. Seems like we need to constrain `scratch_reg` to not be the same as `gp` register. I'll prepare a fix.

udinator, 2020-09-29:
@dariuszst can you apply the fix in #726 and see if this issue is fixed?

dariuszst, 2020-09-30:
@udinator - I've checked some test for different targets (rv32i, rv32imc, rv64imc, 64gc, multi_harts) on the branch with a fix and I didn't notice spike simulation problem as before. So I think the correction of the constraint should solve the problem and will not introduce new ones.

udinator, 2020-09-30:
Thanks @dariuszst! I'll go ahead and close out this issue.

commit 0c640e3a (2020-09-30): Merge pull request #726 from udinator/constrain_gp

fix #725
commit 1b0c8099 (2020-09-30): fix #725
closed 2020-09-30 by udinator