chipsalliance/riscv-dv #742: spike_log_to_trace_csv.py not producing valid CSV

db434, 2020-11-03:
Hi, this is my first time trying to get riscv-dv working, so I'm not sure if this is user error, or a real issue.

I'm trying to use Spike + VCS to generate a trace CSV. The resulting CSV contains two lines per instruction, the second of which is broken. I assume they're both supposed to be merged into a single line.

Here are the steps I've taken:
 1. Compile Spike with `--enable-commitlog`
 1. Generate instructions using VCS
 1. `spike -l --log-commits test.o > test.log 2>&1`
 1. `python3 spike_log_to_trace_csv.py -f --log=test.log --csv=test.csv`

(I've also tried doing steps 2 and 3 through `run.py`, and the output appears identical.)

The log looks sensible, but the CSV looks broken, so I assume the problem is in the Python script. I've attached both, but here's a sample:

Log:
```
core   0: 0x0000000080000016 (0xfff00f1b) addiw   t5, zero, -1
core   0: 3 0x0000000080000016 (0xfff00f1b) x30 0xffffffffffffffff
```

CSV:
```
0000000080000016,addiw,,,fff00f1b,,"addiw   t5, zero, -1","t5,zero,-1",
0000000080000016,x30,,,fff00f1b,,x30 0xffffffffffffffff,0xffffffffffffffff,
```

[test.log](https://github.com/google/riscv-dv/files/5480999/test.log)
[test_csv.txt](https://github.com/google/riscv-dv/files/5481004/test_csv.txt)

hodjat91, 2020-11-03:
Hi @db434 

Since your log file looks okay, it means that all prior steps were successful (instruction generator works as expected, Spike is also properly installed as the output log looks sensible). I tested it and don't see any issue with the script and the generated CSV file looks okay, so here is some suggestions/tips:

1. Make sure that you're working on the most updated branch (master). 
2. Say that you want to generate a random arithmetic program and then convert the output log to CVS. To generate the random program, you can use this command: `python3 run.py --test riscv_arithmetic_basic_test` (ISS&Simulator should be picked up automatically by the script. If not, feel free to specify them in the command). Then, to generate the CSV file you have two options:
a. (preferred) You can get the CSV file as part of getting the functional coverage report. By running this command, you should be able to see the CSV file in `out_2020-11-03/spike_sim/` directory: `python3 cov.py --dir out_2020-11-03/spike_sim/`
b. If you only want to get the CSV file and you don't care about the functional coverage report, you can do so by using `spike_log_to_trace_csv.py` script: `python3 scripts/spike_log_to_trace_csv.py --log riscv_arithmetic_basic_test.0.log --csv csv_name`. 

Either of two should work. Please let us know if you have any further questions. 

-Hodjat

db434, 2020-11-04:
Thanks for the response. I've updated to the latest branch (3467c37), but I'm still seeing the same behaviour with both suggested methods. (Method b actually only outputs a single `ecall` instruction to the csv, but if I add the `-f` flag it behaves the same as method a.)

What further information would be useful to you?

hodjat91, 2020-11-04:
Hi @db434 

I just double checked your spike log, it seems that there is no commit information generated by spike. Then the problem is with the spike installation process. Could you please make sure that you are using the latest version of spike, and you are building it with --enable-commitlog? You might want to install Spike again. Please let us know if the issue persists.

db434, 2020-11-04:
Still the same behaviour with a fresh clone/build of Spike (it wasn't installed either time). I definitely used `--enable-commitlog`.

What should the log look like? The examples I've found by searching look similar to the log I have. Is this not an example of commit information?

```
core   0: 3 0x0000000080000016 (0xfff00f1b) x30 0xffffffffffffffff
```

udinator, 2020-11-07:
Hi @db434 - it seems that the issue here is due to a fairly impactful change made somewhat recently in the Spike repository (that I just found out about).
Some recent changes to Spike have changed how the log files are generated, so our `spike_log_to_trace_csv` script has not been updated to match the new format (which is a real issue).
I'll take a look at and update the scripts as needed when I have some time, but for now I would recommend using OVPsim as your ISS of choice.
Please let me know if you have any other questions.

vskalagi, 2020-11-24:
@db434 the older log files did not contain 'core 0:' part. So there is problem in creating CSV files. Change the spike_log_to_trace_csv.py file to notice this 'core 0:' in every line.

closed 2020-11-29 by db434