chipsalliance/riscv-dv #746: Not able to run a simple hello_world c  program

rahulraveendran15-coder, 2020-11-09:
Hi,
I am not able to run a sample hello world program in the dv instruction generator. Can you please help me on this.
I have added the test in base_testlist.yaml like this

```
- test: riscv_single_c_test
  description: >
     single c test entry
  iterations: 1
  c_test: /home/rr/thesis_ibex/riscv-ovpsim/examples/hello/hello_world.c
```

run command I used is : `run --test riscv_single_c_test --simulator xlm`

When I run this I am getting the error as below what can be the cause for this. I need to generate instructions for hello_world program which is written in c and then I need to put it in ibex core and test
```
Mon, 09 Nov 2020 11:05:16 INFO     Creating output directory: out_2020-11-09
Mon, 09 Nov 2020 11:05:16 INFO     Processing regression test list : /home/rraveendran/thesis_ibex/riscv-dv/target/rv32imc/testlist.yaml, test: riscv_single_c_test
Mon, 09 Nov 2020 11:05:16 INFO     Processing regression test list : /home/rraveendran/thesis_ibex/riscv-dv/yaml/base_testlist.yaml, test: riscv_single_c_test
Mon, 09 Nov 2020 11:05:16 INFO     Found matched tests: riscv_single_c_test, iterations:1
Mon, 09 Nov 2020 11:05:16 INFO     Processing simulator setup file : /home/rraveendran/thesis_ibex/riscv-dv/yaml/simulator.yaml
Mon, 09 Nov 2020 11:05:16 INFO     Found matching simulator: xlm
Mon, 09 Nov 2020 11:05:16 INFO     Building RISC-V instruction generator
Mon, 09 Nov 2020 11:05:17 INFO     Running RISC-V instruction generator
Mon, 09 Nov 2020 11:05:17 INFO     Generating 1 riscv_single_c_test
Mon, 09 Nov 2020 11:05:17 INFO     Running riscv_single_c_test with 1 batches
Traceback (most recent call last):
  File "/home/rraveendran/.local/bin/run", line 11, in <module>
    load_entry_point('riscv-dv', 'console_scripts', 'run')()
  File "/home/rraveendran/thesis_ibex/riscv-dv/run.py", line 1116, in main
    gen(matched_list, args, output_dir, cwd)
  File "/home/rraveendran/thesis_ibex/riscv-dv/run.py", line 388, in gen
    argv.verbose, check_return_code, argv.debug, argv.target)
  File "/home/rraveendran/thesis_ibex/riscv-dv/run.py", line 312, in do_simulate
    output_dir, test['test'])) + \
KeyError: 'gen_test'

```
Also I tried running the hello_world program directly with out adding in test list, at that time it thrown different error
Run command I used: `run --c_tests ../riscv-ovpsim/examples/hello/hello_world.c --simulator xlm`

The error I got is as below:
```
Mon, 09 Nov 2020 11:08:27 INFO     Creating output directory: out_2020-11-09
Mon, 09 Nov 2020 11:08:27 INFO     Compiling c test : ../riscv-ovpsim/examples/hello/hello_world.c
/home/rraveendran/.local/opt/riscv32/riscv-gnu-toolchain/lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: warning: cannot find entry symbol _start; defaulting to 0000000080000000
/home/rraveendran/.local/opt/riscv32/riscv-gnu-toolchain/lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: /tmp/ccJQfldk.o: in function `.L0 ':
hello_world.c:(.text+0x10): undefined reference to `puts'
collect2: error: ld returned 1 exit status
Traceback (most recent call last):
  File "/home/rraveendran/.local/bin/run", line 11, in <module>
    load_entry_point('riscv-dv', 'console_scripts', 'run')()
  File "/home/rraveendran/thesis_ibex/riscv-dv/run.py", line 1020, in main
    args.debug)
  File "/home/rraveendran/thesis_ibex/riscv-dv/run.py", line 575, in run_c
    run_cmd_output(cmd.split(), debug_cmd=debug_cmd)
  File "/home/rraveendran/thesis_ibex/riscv-dv/scripts/lib.py", line 195, in run_cmd_output
    raise exc
  File "/home/rraveendran/thesis_ibex/riscv-dv/scripts/lib.py", line 192, in run_cmd_output
    output = subprocess.check_output(cmd)
  File "/usr/lib64/python3.6/subprocess.py", line 356, in check_output
    **kwargs).stdout
  File "/usr/lib64/python3.6/subprocess.py", line 438, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['/home/rraveendran/.local/opt/riscv32/riscv-gnu-toolchain//bin/riscv32-unknown-elf-gcc', '-mcmodel=medany', '-nostdlib', '-nostartfiles', '../riscv-ovpsim/examples/hello/hello_world.c', '-I/home/rraveendran/thesis_ibex/riscv-dv/user_extension', '-T/home/rraveendran/thesis_ibex/riscv-dv/scripts/link.ld', '-o', 'out_2020-11-09/directed_c_tests/hello_world.o', '-march=rv32imc', '-mabi=ilp32']' returned non-zero exit status 1.
```
What is the difference between both the approaches I made adding c test in test list and directly running what is the difference?
Appreciate any help, thanks in advance

taoliug, 2020-11-30:
I fixed the script and the doc, now you should be able use c_test both in command line and yaml testlist to specify a C test. However, in order to get the C test properly compiled and simulated, more things need to be done, like trap handling, end of test communication with ISS etc. I will add some sample C test soon. For the last error you showed, please make sure the related library is properly compiled along with the C program. You can use --gcc_opts to add additional compile options as needed.

closed 2020-11-30 by taoliug

franAyachi, 2025-02-25:
Hi! is there any way to also generate new C_test or assembly between the main program generated by riscv-dv? like introduce directed tests randomly inside the riscv-dv generated test