chipsalliance/riscv-dv #749: The coverage of A extension added by myself is invalid

blackcatruri, 2020-11-10:
Hello，
Dv lacks coverage of A extension, right? So i try to add it. But it seems that dv doesn't collect the coverage  in the result. Can you give me some advice? In addition, will you add this feature later?

udinator, 2020-11-11:
Hi @blackcatruri, I think it would be useful if you could provide a pointer to the coverage implementation that you have created to get a better idea of what is going wrong.
Adding @weicaiyang to the discussion as well for any other inputs.

blackcatruri, 2020-11-12:
Hi @weicaiyang, 
I tried to add the following code in the corresponding location of riscv_instr_cover_group.sv file :
covergroup amoadd_w_cg with function sample(riscv_instr instr);
  cp_rd             : coverpoint instr.rd;
  cp_rs1            : coverpoint instr.rs1;
  cp_rs2            : coverpoint instr.rs2;
endgroup

amoadd_w_cg = new()；

AMOADD_W       :  `SAMPLE(amoadd_w_cg, instr)

and I deleted the following part of riscv_instr_cov_test.sv file: 
            if (uvm_is_match("amo*",trace["instr"]) ||          
                uvm_is_match("lr*" ,trace["instr"]) ||
                uvm_is_match("sc*" ,trace["instr"])) begin
              // TODO: Enable functional coverage for AMO test
              continue;

However, the coverage of the amoadd_w instruction is 0, even if there is corresponding code in riscv_amo_test.s file.
Did I miss something? 
Thanks.

weicaiyang, 2020-11-12:
> Hi @weicaiyang,
> I tried to add the following code in the corresponding location of riscv_instr_cover_group.sv file :
> covergroup amoadd_w_cg with function sample(riscv_instr instr);
> cp_rd : [...]

Thanks for adding this. Have you added R32A and R64A here? 
https://github.com/google/riscv-dv/blob/3da32bbf6080d3bf252a7f71c5e3a32ea4924e49/test/riscv_instr_cov_test.sv#L134

You can also try to add print `$display(instr.instr_name); ` before this line to see if any AMO instruction gets to this point.
https://github.com/google/riscv-dv/blob/3da32bbf6080d3bf252a7f71c5e3a32ea4924e49/test/riscv_instr_cov_test.sv#L139

If it's still not working, could you push your code as draft PR, so that we can take a look? Thanks

blackcatruri, 2020-11-16:
@weicaiyang  Thanks for your advice,but it doesn’t seem to work.The reason seems to be that the log to csv script had a problem converting the A extension instruction.The operant item in the csv file is missing the rd register information.I have uploaded my code and csv file.
[test.zip](https://github.com/google/riscv-dv/files/5543638/test.zip)

blackcatruri, 2020-11-16:
Forgot to say, when matching lr.w, lr.d, sc.w, sc.d, UVM_FATAL will be reported, reason unknown.

weicaiyang, 2020-11-16:
The UVM_FATAL may be from below block which parses the operand values from csv file.
These instructions are R_FORMAT, but the format isn't supported in the base class.
https://github.com/google/riscv-dv/blob/f64243fabc79672f33b0e966dac6e9dcee5e861e/src/isa/riscv_instr_cov.svh#L308

We need to override the `update_src_regs` function in [riscv_amo_instr.sv](https://github.com/google/riscv-dv/blob/master/src/isa/riscv_amo_instr.sv). b-ext is perhaps a good example. You can follow it to make the update. 
https://github.com/google/riscv-dv/blob/f64243fabc79672f33b0e966dac6e9dcee5e861e/src/isa/riscv_b_instr.sv#L501

blackcatruri, 2020-11-17:
Ok, thanks, I’ll try.

taoliug, 2021-03-22:
Closing some stale bug because of lack of activity. Please open new bug if it's still an issue

closed 2021-03-22 by taoliug