chipsalliance/riscv-dv #750: Error: unrecognized opcode `fsrmi 4'

suhasskrishanmurthy, 2020-11-13:
Hi,
  I am trying to run floating point test cases with rv64imafd arch by using following command.
**python3 run.py --custom_target ./target/rv64imafd --isa rv64imafd --mabi lp64d --test riscv_floating_point_arithmetic_test  --iterations 5 --simulator questa**

But it showing error as shown below:

Fri, 13 Nov 2020 14:53:03 INFO     Creating output directory: out_2020-11-13
Fri, 13 Nov 2020 14:53:03 INFO     Processing regression test list : ./target/rv64imafd/testlist.yaml, test: riscv_floating_point_arithmetic_test
Fri, 13 Nov 2020 14:53:03 INFO     Processing regression test list : /home/ariane/tandem/test_vectors/riscv-dv/target/rv64imc/testlist.yaml, test: riscv_floating_point_arithmetic_test
Fri, 13 Nov 2020 14:53:03 INFO     Processing regression test list : /home/ariane/tandem/test_vectors/riscv-dv/target/rv32imc/testlist.yaml, test: riscv_floating_point_arithmetic_test
Fri, 13 Nov 2020 14:53:03 INFO     Processing regression test list : /home/ariane/tandem/test_vectors/riscv-dv/yaml/base_testlist.yaml, test: riscv_floating_point_arithmetic_test
Fri, 13 Nov 2020 14:53:03 INFO     Found matched tests: riscv_floating_point_arithmetic_test, iterations:5
Fri, 13 Nov 2020 14:53:03 INFO     Processing simulator setup file : /home/ariane/tandem/test_vectors/riscv-dv/yaml/simulator.yaml
Fri, 13 Nov 2020 14:53:03 INFO     Found matching simulator: questa
Fri, 13 Nov 2020 14:53:03 INFO     Building RISC-V instruction generator
Fri, 13 Nov 2020 14:53:28 INFO     Running RISC-V instruction generator
Fri, 13 Nov 2020 14:53:28 INFO     Generating 5 riscv_floating_point_arithmetic_test
Fri, 13 Nov 2020 14:53:28 INFO     Running riscv_floating_point_arithmetic_test with 1 batches
Fri, 13 Nov 2020 14:53:28 INFO     Running riscv_floating_point_arithmetic_test, batch 1/1, test_cnt:5
Fri, 13 Nov 2020 14:55:00 INFO     Compiling out_2020-11-13/asm_tests/riscv_floating_point_arithmetic_test_0.S
out_2020-11-13/asm_tests/riscv_floating_point_arithmetic_test_0.S: Assembler messages:
out_2020-11-13/asm_tests/riscv_floating_point_arithmetic_test_0.S:182: **Error: unrecognized opcode `fsrmi 4'**
Traceback (most recent call last):
  File "run.py", line 1144, in <module>
    main()
  File "run.py", line 1122, in main
    args.gcc_opts, args.debug)
  File "run.py", line 435, in gcc_compile
    run_cmd_output(cmd.split(), debug_cmd=debug_cmd)
  File "/home/ariane/tandem/test_vectors/riscv-dv/scripts/lib.py", line 195, in run_cmd_output
    raise exc
  File "/home/ariane/tandem/test_vectors/riscv-dv/scripts/lib.py", line 192, in run_cmd_output
    output = subprocess.check_output(cmd)
  File "/usr/lib/python3.6/subprocess.py", line 356, in check_output
    **kwargs).stdout
  File "/usr/lib/python3.6/subprocess.py", line 438, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['/home/ariane/tandem/toolchain/bin/riscv64-unknown-elf-gcc', '-static', '-mcmodel=medany', '-fvisibility=hidden', '-nostdlib', '-nostartfiles', 'out_2020-11-13/asm_tests/riscv_floating_point_arithmetic_test_0.S', '-I/home/ariane/tandem/test_vectors/riscv-dv/user_extension', '-T/home/ariane/tandem/test_vectors/riscv-dv/scripts/link.ld', '-o', 'out_2020-11-13/asm_tests/riscv_floating_point_arithmetic_test_0.riscv', '-march=rv64imafd', '-mabi=lp64d']' returned non-zero exit status 1.

Please Tell me Where I am going wrong.
Thanks in advance.

taoliug, 2020-11-29:
It seems working for me, can you run with "-v" and share the log?

python3 run.py --custom_target ./target/custom_1 --isa rv64imafd --mabi lp64d --test riscv_floating_point_arithmetic_test
Sun, 29 Nov 2020 15:50:57 INFO     Creating output directory: out_2020-11-29
Sun, 29 Nov 2020 15:50:57 INFO     Processing regression test list : ./target/custom_1/testlist.yaml, test: riscv_floating_point_arithmetic_test
Sun, 29 Nov 2020 15:50:57 INFO     Processing regression test list : /usr/local/google/home/taliu/riscv-dv/target/rv64imc/testlist.yaml, test: riscv_floating_point_arithmetic_test
Sun, 29 Nov 2020 15:50:57 INFO     Processing regression test list : /usr/local/google/home/taliu/riscv-dv/target/rv32imc/testlist.yaml, test: riscv_floating_point_arithmetic_test
Sun, 29 Nov 2020 15:50:57 INFO     Processing regression test list : /usr/local/google/home/taliu/riscv-dv/yaml/base_testlist.yaml, test: riscv_floating_point_arithmetic_test
Sun, 29 Nov 2020 15:50:57 INFO     Found matched tests: riscv_floating_point_arithmetic_test, iterations:1
Sun, 29 Nov 2020 15:50:57 INFO     Processing simulator setup file : /usr/local/google/home/taliu/riscv-dv/yaml/simulator.yaml
Sun, 29 Nov 2020 15:50:57 INFO     Found matching simulator: vcs
Sun, 29 Nov 2020 15:50:57 INFO     Building RISC-V instruction generator
Sun, 29 Nov 2020 15:51:01 INFO     Running RISC-V instruction generator
Sun, 29 Nov 2020 15:51:01 INFO     Generating 1 riscv_floating_point_arithmetic_test
Sun, 29 Nov 2020 15:51:01 INFO     Running riscv_floating_point_arithmetic_test with 1 batches
Sun, 29 Nov 2020 15:51:01 INFO     Running riscv_floating_point_arithmetic_test, batch 1/1, test_cnt:1
Sun, 29 Nov 2020 15:51:05 INFO     Compiling out_2020-11-29/asm_tests/riscv_floating_point_arithmetic_test_0.S
Sun, 29 Nov 2020 15:51:05 INFO     Converting to out_2020-11-29/asm_tests/riscv_floating_point_arithmetic_test_0.bin
Sun, 29 Nov 2020 15:51:05 INFO     Processing ISS setup file : /usr/local/google/home/taliu/riscv-dv/yaml/iss.yaml
Sun, 29 Nov 2020 15:51:05 INFO     Found matching ISS: spike
Sun, 29 Nov 2020 15:51:05 INFO     spike sim log dir: out_2020-11-29/spike_sim
Sun, 29 Nov 2020 15:51:05 INFO     Running spike sim: out_2020-11-29/asm_tests/riscv_floating_point_arithmetic_test_0.o

Chakri-myneni, 2020-12-01:
@suhasskrishanmurthy  It is the issue with version of riscv tool chain. please check it.

suhasskrishanmurthy, 2020-12-01:
Hi @Chakri-myneni , I looked and my tool-chain was 7.2.0 and got to know that this version of tool-chain is broken. So installed fresh tool-chain of version 9.2.0 and the issue got resolved.

taoliug, 2020-12-01:
Thanks for the update, closing the issue now

closed 2020-12-01 by taoliug