chipsalliance/riscv-dv #759: run riscv-dv with any other simulator except mentioned

muhammadsalmanafzal, 2020-12-03:
Hey there, I was hoping to ask if there is any way to run riscv-dv without the tools mentioned in the pre-req since I'm a student and I don't have any access to any of the tools. I'll appreciate it if someone can suggest a workaround.

RojasMilo, 2020-12-07:
Hi @muhammadsalmanafzal I tried with:

run --test riscv_arithmetic_basic_test --target=rv32i --steps gen --simulator **pyflow** --verbose

To generate a test with python flow, but only work onces :( (generate an assembly program) then I have timeouts:

Mon, 07 Dec 2020 11:00:27 lib.py:121   ERROR    Timeout[1000s]:

Mon, 07 Dec 2020 11:00:27 lib.py:131   DEBUG

udinator, 2020-12-11:
Hi @muhammadsalmanafzal, like @RojasMilo has suggested, you can try using the `--simulator pyflow` to enable the Python-based riscv-dv generator (please note this is under active development).

@hodjat91 @aneels3 - are you guys experiencing any timeouts when running the pyflow?

aneels3, 2020-12-11:
Hi @udinator 
Yes, I am experiencing the timeouts. like you said it is under active development so it is not performance-optimized yet. 
@muhammadsalmanafzal  you can run the python generator as suggested by @RojasMilo and @udinator.
We can make it work by doing some minor changes in the run command. Please add the --gen_timeout knob for now.

Try using the command like this.
`python3 run.py --test riscv_arithmetic_basic_test --target=rv32i --steps gen --simulator pyflow --gen_timeout=1000`
maybe some higher value for gen_timeout (If you are running multiple iterations)

or you can decrease the number of instructions from the base_teslist.yml file maybe 1k or 2k.

Let me know if you are experiencing any further issues.
Regards,
Anil

taoliug, 2021-03-22:
Closing this bug as there's a bug tracking pyflow

closed 2021-03-22 by taoliug