chipsalliance/riscv-dv #760: Floating point instructions: How to control rounding mode?

agrobman, 2020-12-08:
Hi,

I need to limit rounding modes for some FP instructions. How do I do this?
where is in the code the rounding mode  added?

Thanks in advance
Alex

taoliug, 2021-01-07:
Here's a global random knob to control the rounding mode. It's applied globally to all FP instructions. Please let me know if it meets your requirement.

https://github.com/google/riscv-dv/blob/master/src/riscv_instr_gen_config.sv#L78

agrobman, 2021-01-07:
I saw this, but FP opcodes also include RM field, how does riscv-dv generate such instructions with "static" RM.

assembler allows something like:
fadd  fx, fy, fz, rmm

taoliug, 2021-01-08:
Sure, I will add this support shortly

taoliug, 2021-01-08:
This feature is added, PTAL if it works for you.

agrobman, 2021-01-08:
it seems rm is added for all FP instructions test  - we need some, maybe 50% of instructions, taking rm from fcsr... 

I got it with :

```
diff --git a/src/isa/riscv_floating_point_instr.sv b/src/isa/riscv_floating_point_instr.sv
index e65c5b1..4ecaa31 100644
--- a/src/isa/riscv_floating_point_instr.sv
+++ b/src/isa/riscv_floating_point_instr.sv
@@ -20,6 +20,8 @@ class riscv_floating_point_instr extends riscv_instr;
   rand riscv_fpr_t fs2;
   rand riscv_fpr_t fs3;
   rand riscv_fpr_t fd;
+  rand f_rounding_mode_t rm;
+  rand bit         static_rm;
   bit              has_fs1 = 1'b1;
   bit              has_fs2 = 1'b1;
   bit              has_fs3 = 1'b0;
@@ -67,6 +69,10 @@ class riscv_floating_point_instr extends riscv_instr;
       default:
         `uvm_fatal(`gfn, $sformatf("Unsupported floating point format: %0s", format.name()))
     endcase
+    if(format == R4_FORMAT || instr_name inside {FADD_S, FSUB_S, FMUL_S, FDIV_S, FSQRT_S, FCVT_W_S, FCVT_WU_S, FCVT_S_W, FCVT_S_WU,
+                                                 FADD_D, FSUB_D, FMUL_D, FDIV_D, FSQRT_D, FCVT_D_S, FCVT_S_D,  FCVT_D_W, FCVT_D_WU,
+                                                 FCVT_W_D, FCVT_WU_D})
+        asm_str = {asm_str, static_rm ? {", ", rm.name()} : ""};
     if(comment != "")
       asm_str = {asm_str, " #",comment};
     return asm_str.tolower();
@@ -84,6 +90,8 @@ class riscv_floating_point_instr extends riscv_instr;
     this.has_fs2 = rhs_.has_fs2;
     this.has_fs1 = rhs_.has_fs1;
     this.has_fd  = rhs_.has_fd;
+    this.rm      = rhs_.rm;
+    this.static_rm  = rhs_.static_rm;
   endfunction : do_copy

```

taoliug, 2021-01-08:
Add another knob for it, this should give you 50% of chance picking the rounding mode from the instruction.

closed 2021-01-08 by taoliug