chipsalliance/riscv-dv #761: FPU L/S instructions data sections.

agrobman, 2020-12-08:
Hi,

Where are the FP load/stores  targeted to? What are the data patterns for them?
Are these valid FP numbers or just any random patterns? Are these data sections shared with fixed L/S instructions?

When random program is executed the FP regs have tendency to saturation - to become NaNs or 0 etc.
Does riscv-dv "refresh" FPU regs periodically with valid FP numbers?

Thanks in advance
Alex

taoliug, 2021-01-07:
We currently have a routine to initialize the floating GPR to some interesting values 
https://github.com/google/riscv-dv/blob/master/src/riscv_asm_program_gen.sv#L641

This doesn't apply to data sections. It's done at the beginning of the program. We can make it randomly injected in the program to "refresh FPU"

agrobman, 2021-01-07:
there is one more problem - fflags are also saturate with time, reducing their testing efficiency.
Also can the RM in fcsr be selected/updated randomly multiple times in the test, including a small percentage of illegal values?