chipsalliance/riscv-dv #770: importlib.metadata.PackageNotFoundError: riscv-dv

panda1628, 2021-01-18:
It is more a question than an issue

I install riscv-dv and I can run it, but other users run into this issue by running the same command.

Any idea how to fix it?

/home/xxx/.local/bin/run --test riscv_arithmetic_basic_test --simulator ius -i 1 --target rv32i

Getting this error
Traceback (most recent call last):
  File "/home/xxx/.local/bin/run", line 33, in <module>
    sys.exit(load_entry_point('riscv-dv', 'console_scripts', 'run')())
  File "/home/xxx/.local/bin/run", line 22, in importlib_load_entry_point
    for entry_point in distribution(dist_name).entry_points
  File "/share/common/pkgs/python/v3.9.0/lib/python3.9/importlib/metadata.py", line 524, in distribution
    return Distribution.from_name(distribution_name)
  File "/share/common/pkgs/python/v3.9.0/lib/python3.9/importlib/metadata.py", line 187, in from_name
    raise PackageNotFoundError(name)
importlib.metadata.PackageNotFoundError: riscv-dv

taoliug, 2021-03-22:
Please try 

python3 run.py --test riscv_arithmetic_basic_test --simulator ius -i 1 --target rv32i

closed 2021-03-22 by taoliug

panda1628, 2021-03-22:
[xxx@hw-sh riscv-dv]$ python3 run.py --test riscv_arithmetic_basic_test --simulator ius -i 1 --target rv32i --isa=rv32ima --mabi ilp32
Mon, 22 Mar 2021 00:23:04 INFO     Creating output directory: out_2021-03-22
Mon, 22 Mar 2021 00:23:04 INFO     Processing regression test list : /fac/proj_xxx/xxx/riscv-dv/target/rv32i/testlist.yaml, test: riscv_arithmetic_basic_test
Traceback (most recent call last):
  File "/fac/proj_xxx/xxx/riscv-dv/scripts/lib.py", line 63, in read_yaml
    yaml_data = yaml.safe_load(f)
AttributeError: module 'yaml' has no attribute 'safe_load'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "run.py", line 1144, in <module>
    main()
  File "run.py", line 1036, in main
    matched_list, cwd)
  File "/fac/proj_xxx/xxx/riscv-dv/scripts/lib.py", line 216, in process_regression_list
    yaml_data = read_yaml(testlist)
  File "/fac/proj_xxx/xxx/riscv-dv/scripts/lib.py", line 64, in read_yaml
    except yaml.YAMLError as exc:
AttributeError: module 'yaml' has no attribute 'YAMLError'

Do I need to git pull the latest? Thx