chipsalliance/riscv-dv #781: Supported tests from testlist in pyflow

amnafayyaz-lm, 2021-02-23:
I was working on enabling pyflow in google's riscv_dv flow for that I was able to generate riscv_arithmetic_basic_test from the testlist using the run.py script. In riscv_arithmetic_basic_test case assembly test files are successfully generated. But I am facing issues in generation of other random tests in the testlist e.g. riscv_rand_instr_test, riscv_jump_stress_test, riscv_loop_test etc. Kindly confirm which tests from testlist are supported in pyflow. I am facing the issue shown below while generating the others.

**ERROR return code: True/1, cmd: python3 /home/users/scratch/amnafayyaz.data/riscv-dv/pygen/pygen_src/test/riscv_instr_base_test.py --num_of_tests=2 --start_idx=0 --asm_file_name=out_2021-02-23/asm_test/riscv_loop_test --log_file_name=out_2021-02-23/sim_riscv_loop_test_0.log  --target=rv32imc --instr_cnt=10000 --num_of_sub_program=5 --directed_instr_1=riscv_loop_instr,20**

I am using the folowing comand to run the script:
python3 run.py --test riscv_loop_test --target=rv32imc --steps gen --simulator pyflow --gen_timeout=10000

aneels3, 2021-02-23:
Hi @amnafayyaz-lm 
Thanks for letting us know about it.
Only a few selected tests are available for pyflow as it is in the development phase currently.
Please find the supported tests below.

1. riscv_arithmetic_basic_test
2. riscv_amo_test (--target rv32imafdc)
3. riscv_floating_point_arithmetic_test
4. riscv_b_ext_test

- Directed tests

1. riscv_int_numeric_corner_stream
2. riscv_jal_instr
3. riscv_amo_instr_stream

- Available Extensions
1. RV32I
2. RV32IMC
3. RV32IMCB
4. RV32IMAFDC

Hope this helps.

Regards,
Anil

amnafayyaz-lm, 2021-02-23:
Yes it was helpful. Thank you.

fPecc, 2022-07-29:
Hi,

Are there more tests supported by pyflow now?