chipsalliance/riscv-dv #790: empty grp field w/ ovpsim and instr_trace_compare mismatches

mescoba1, 2021-03-31:
the ovpsim trace functionality does not record the gpr's for all instructions executed (i.e. `Info 48:`, `Info 49:`)
```
Info 47: 'riscvOVPsim/cpu', 0x00000000000000b8(main): Machine 00eb8d33 add     s10,s7,a4                                                                                                                                                                  
Info   s10 80000000 -> 00000001                                                                                                                                                                                                                           
Info 48: 'riscvOVPsim/cpu', 0x00000000000000bc(main+4): Machine 03bbcc63 blt     s7,s11,f4                                                                                                                                                                
Info 49: 'riscvOVPsim/cpu', 0x00000000000000c0(main+8): Machine 003322b3 slt     t0,t1,gp                                                                                                                                                                 
Info 50: 'riscvOVPsim/cpu', 0x00000000000000c4(main+c): Machine e9ea0593 addi    a1,s4,-354                                                                                                                                                               
Info   a1 915ae93b -> fffffe9e
  ```
  
when the ovpsim log gets converted to a csv file, the gpr column will be empty for the gpr's that were not recorded within the log
```
00000000000000b8,add,s10:00000001,,00eb8d33,3,"add     s10,s7,a4","s10,s7,a4",                                                                                                                                                                            
00000000000000bc,blt,,,03bbcc63,3,"blt     s7,s11,f4","s7,s11,0x38",                                                                                                                                                                                      
00000000000000c0,slt,,,003322b3,3,"slt     t0,t1,gp","t0,t1,gp",                                                                                                                                                                                          
00000000000000c4,addi,a1:fffffe9e,,e9ea0593,3,"addi    a1,s4,-354","a1,s4,-354",                                                                                                                                                                          
00000000000000c8,auipc,a4:a07cc0c8,,a07cc717,3,"auipc   a4,0xa07cc","a4,0xa07cc",
  ```
  
the trace `instr_trace_compare.py` script, however, will still attempt to compare all the gpr's within the two csv files. this leads to cascading mismatches where it attempts to compare GPRs from two different instructions entirely (instruction at `0xbc` against `0xc4`)
```
Mismatch[1]:
ovpsim[47] : pc[00000000000000c4] addi    a1,s4,-354: a1:fffffe9e 
ibex[47] : pc[000000bc] blt      x23,x27,f4: s8:00000000 
```

the ibex tracer stores and populates the GPRs for the instructions at `0xbc`, `0xc0`, and `0xc4`. are there ways to mitigate this problem within `instr_trace_compare.py` script? the only solution I can think of, is maybe passing `compare_final_value_only=1` into `compare_trace_csv` invocation?

eroom1966, 2021-03-31:
@mescoba1 
The trace mode being used reports register changes, not their values 
If no register values have changed in this instruction execution, then there is nothing to report 
can you confirm whether any registers were changed by this instruction?
If so that would require investigation 
Thx
Lee

mescoba1, 2021-03-31:
I crossed reference with the RTL, looks like the register was not being changed, however, the RTL tracing module still recorded the change (ignored write enable to the register file). Traces matched after updating the trace module connections.

Thanks for the help!