chipsalliance/riscv-dv #793: Store segfault riscv_load_store_rand_addr_instr_stream

mescoba1, 2021-05-10:
When running a randomized load store test with `riscv_load_store_rand_addr_instr_stream`, the ISS will throw an exception, saying that we are trying to access an invalid memory region.
```
 27 Info 120: 'riscvOVPsim/cpu', 0x00000000000001dc(main+138): Machine 221b9123 sh      ra,546(s7)
 26 Processor Exception (PC_PRX) Processor 'riscvOVPsim/cpu' 0x1dc: 221b9123 sh      ra,546(s7)
 25 Processor Exception (PC_WPX) No write access at 0xa11f526a 
 ```
 
 my memory map is defined as follows:
 ```
 --memory imem:0x00000000:0x00008000:rx,dmem:0x80000000:0x80008000:rw
 ```

in the generator we define the memory regions and set their physical address with the linker script.
```   
    // Memory Region Sizing (link.ld)
    cfg.stack_len = 'h400;
    cfg.mem_region = '{
      '{name:"imem",   size_in_bytes: 'h00002000, xwr: 3'b101},
      '{name:"dmem",   size_in_bytes: 'h00002000, xwr: 3'b011}
    };
```

in the assembly file, we look to be loading a large immediate value of `0x211f3000` (outside the range of our 32KB memory at 0x8000000 when added w/ base address register `s7`)
```
                  la         s7, dmem+0 #start riscv_load_store_rand_addr_instr_stream_7
                  li         a0, 0x211f3000
                  add        s7, s7, a0
                  sh         ra, 546(s7)
 ```
 
 disassembly:
 ```
     1d0:	e7cb8b93          	addi	s7,s7,-388 # 80002048 <__stack_size+0xffffcc00>
     1d4:	211f3537          	lui	a0,0x211f3
     1d8:	00ab8bb3          	add	s7,s7,a0
     1dc:	221b9123          	sh	ra,546(s7)
 ```
 
 the immediate value for this instruction stream does not look to be constrained when randomizing the `li` instruction [here](https://github.com/google/riscv-dv/blob/master/src/riscv_load_store_instr_lib.sv#L473-L479)
 
 ```
     // Use LI to initialize the address offset
    li_instr = riscv_pseudo_instr::type_id::create("li_instr");
    `DV_CHECK_RANDOMIZE_WITH_FATAL(li_instr,
       pseudo_instr_name == LI;
       rd inside {cfg.gpr};
       rd != gpr;
    )
 ```
 
 shouldn't the `riscv_load_store_rand_addr_instr_stream` instruction stream take the memory range into account when randomizing the immediate value for the address offset? Otherwise you run the risk of exceeding the memory region.

mescoba1, 2021-05-11:
Maybe we should be constraining the `li_instr.imm` value for a given memory size and base
```
// riscv_load_store_rand_addr_instr_stream
  virtual function void add_rs1_init_la_instr(riscv_reg_t gpr, int id, int base = 0);
  ...
    // Use LI to initialize the address offset
    li_instr = riscv_pseudo_instr::type_id::create("li_instr");
    `DV_CHECK_RANDOMIZE_WITH_FATAL(li_instr,
       pseudo_instr_name == LI;
       rd inside {cfg.gpr};
       rd != gpr;
       imm < (cfg.mem_region[id].size_in_bytes - base);
    )
    ...
    super.add_rs1_init_la_instr(gpr, id, 0); // riscv_mem_access_stream
  endfunction
  ```

mcockrell-google, 2021-06-07:
Hi Miguel - can you join our riscvdv CHIPS Alliance Workgroup. We meet once a month to discuss such issues.

https://lists.chipsalliance.org/g/riscv-dv-wg

Thanks - Matt