chipsalliance/riscv-dv #796: "illegal operands" and "already defined symbol" in generated .S test

mykola-mmm, 2021-05-20:
Hi everyone,

The version I'm working with: `commit 5a79a54f8330c904eae98e51be55bbb55ac30c82`
Here is my riscv_core_setting.sv:

> // Implemented previlieged CSR list
> `ifdef DSIM
> privileged_reg_t implemented_csr[] = {
> `else
> const privileged_reg_t implemented_csr[] = {
> `endif
>     // Machine mode mode CSR
>     MSTATUS,    // Machine status
>     MISA,       // ISA and extensions
>     MTVEC,      // Machine trap-handler base address
>     MEPC,       // Machine exception program counter
>     MCAUSE     // Machine trap cause
> };

And here is base_testlist.yaml with my custom test:

> - test: sample_jump
>   description: >
>     Arithmetic instruction test, no load/store/branch instructions
>   gen_opts: >
>     +num_of_tests=5
>     +num_of_sub_program=5
>     +instr_cnt=1000
>     +directed_instr_0=riscv_jump_instr,8
>     +no_csr_instr=1
>   iterations: 10
>   gen_test: riscv_instr_base_test
>   rtl_test: core_base_test

The errors I`m getting:

> out_2021-05-20/asm_test/sample_jump_0.S: Assembler messages:
> out_2021-05-20/asm_test/sample_jump_0.S:307: Error: illegal operands `la t1,'
> out_2021-05-20/asm_test/sample_jump_0.S:317: Error: symbol `_j0' is already defined
> out_2021-05-20/asm_test/sample_jump_0.S:579: Error: illegal operands `la s5,'
> out_2021-05-20/asm_test/sample_jump_0.S:586: Error: symbol `_j0' is already defined
> out_2021-05-20/asm_test/sample_jump_0.S:845: Error: illegal operands `la a3,'
> out_2021-05-20/asm_test/sample_jump_0.S:856: Error: symbol `_j0' is already defined
> out_2021-05-20/asm_test/sample_jump_0.S:1127: Error: illegal operands `la t3,'
> out_2021-05-20/asm_test/sample_jump_0.S:1136: Error: symbol `_j0' is already defined

Could you please point out what am I doing wrong here?

UrvishkumarPatel, 2021-09-03:
I am also facing illegal operand error in one of the testcase.

anwesh-27, 2022-07-22:
Hi,
I have solved "already defined symbol" issue and currently working on "illegal operands".
Thank you