chipsalliance/riscv-dv #801: RV32E instruction set support

mescoba1, 2021-05-26:
Is RV32E a supported instruction set? I am aware that `NUM_GPR` can be configured within `riscv_core_settings.sv`:
```
parameter int NUM_GPR = 16;     
```

However, upon doing so, registers 16-32 will still be reference, and GCC will complain about those illegal registers being reference when compiled with the `march=rv32e`.

mykola-mmm, 2021-05-27:
Hi!

Could you try adding "-fpermissive" flag to the "gcc_opts"?
Not sure this would help but definetly worth trying.

ankurcere, 2024-03-19:
I've been experiencing the same issue described here and I'm curious to know if there have been any updates regarding its resolution. Could someone kindly provide an update on the status of this issue?

It seems like quite a few users are encountering this problem, so any information on progress or potential workarounds would be greatly appreciated.

zeeshanrafique23, 2024-03-19:
Try compiling with GCC flag -mabi=ilp32e
Read further here: https://gcc.gnu.org/onlinedocs/gcc/RISC-V-Options.html

ankurcere, 2024-03-21:
I tried this but it didn't work in my case.
I'm currently facing an issue with limiting the number of registers used in my RISC-V assembly code (.S file) while generating the code. Despite setting NUM_GPR=16, the code still utilizes registers beyond x0 to x15. I need to restrict the code to only use registers x0 to x15 throughout its execution. Is RV32E supported in RISCV-DV.
I've tried setting NUM_GPR=16, but it appears to regulate only the init general purpose registers with random values. Later on, it uses other registers also.
Thank you in advance for any assistance or guidance provided.