chipsalliance/riscv-dv #805: cannot generate riscv_csr_test.

suhasskrishanmurthy, 2021-07-06:
Hi,
  I am trying to generate **riscv_csr_test**. using the 
cmd :- **python3 run.py --custom_target ./target/rv64imafd --isa rv64imafd --mabi lp64d --test riscv_csr_test  --simulator questa**

it is showing error stating: 
Cannot find riscv_csr_test in ./target/rv64imafd/testlist.yaml

But the testcase case is present in **./yaml/base_testlist.yaml** 
and also this issue is not arising for the testcase **riscv_arithmetic_basic_test** even though it is not in **./target/rv64imafd/testlist.yaml.**

**python3 run.py --custom_target ./target/rv64imafd --isa rv64imafd --mabi lp64d --test riscv_arithmetic_basic_test --simulator questa**

Please can anyone help me in resolving this issue.
Thanks in advance.

hodjat91, 2021-07-07:
@aneels3 Are we supporting rv64imafd ISA at this point? Could you PTAL into this issue? 

Thanks,
Hodjat

aneels3, 2021-07-08:
Hi @hodjat91 
We do not have rv64imafd instead we have rv32imafdc.
I think @suhasskrishanmurthy is trying to run his own custom target.

I have not run the csr_test test yet, but looking at the [csr_test](https://github.com/google/riscv-dv/blob/2b2abda6fa0fdd35435cca615861fe3145996b14/yaml/base_testlist.yaml#L176) in base_testlist.yaml, I don't see any gen_test for it.

suhasskrishanmurthy, 2021-07-09:
Thanks for replying.
yes I am using a custom target. but it is the same issue for rv64gc.
@aneels3 how to add gen_test for riscv_csr_test?