chipsalliance/riscv-dv #806: Infinite looping tests generated when PMP is enabled with user/supervisor mode

sammy17, 2021-07-09:
Generated test with the following configuration resulted in an infinite looping testcase. 
I encountered many such looping tests when I enabled PMP. 
Seems like any test with PMP enabled and uses supervisor mode or user mode faces this issue. 

The first instruction fault exception happens when the program tries to execute first user mode or supervisor mode instruction, after that it keeps looping. 

I noticed that such PMP exceptions should be handled by the PMP exception routine generated by the `gen_pmp_exception_routine` function to avoid infinite loops. But PMP exception routine keeps writing 15 (0xf) to the pmpcfg0 CSR and the same exception happens when trying to execute the smode or umode instruction again. 

One of the tests that I encountered this issue is `riscv_rand_jump_test	`with SEED **894253563**. 

I have attached the assembly file, binary, disassembly, and spike trace log. 

Really appreciate any help to identify and fix this issue. 

[test.zip](https://github.com/google/riscv-dv/files/6794265/test.zip)

sammy17, 2021-07-22:
I am seeing this issue in the latest riscv-dv version as well.

**You can easily reproduce this issue with the following command:**
`python3 run.py --steps gen,gcc_compile,iss_sim --iss spike --simulator vcs --iterations 1 -tn riscv_rand_jump_test --isa rv64gc --mabi lp64 --seed 894253563 --target rv64gc`
 
I am using VCS Q-2020.03 as the simulator. 
Any comments, feedback, or directions on how to resolve this issue are appreciated.

somyadashora, 2021-09-02:
Yes, we also encountered the similar issue where the instruction access fault was raised and `mmode_exception_handler `routine was called. When returning from routing using mret again the exception (instruction fault was raised). The spike ISS was stuck in this loop forever. The exception handling routine writes the same value to pmpcfg0 register as before.

The test we ran was riscv_pmp_test with following config:
```
- test: riscv_pmp_test
  description: >
    Provide some PMP configuration parameters, and setup PMP CSRs appropriately
  iterations: 1
  gen_test: riscv_rand_instr_test
  gen_opts: >
    +pmp_granularity=0
    +pmp_num_regions=4
    +pmp_randomize=1
    +pmp_max_offset=0x3fffffff
    +pmp_allow_addr_overlap=1
    +enable_write_pmp_csr=1
  rtl_test: core_base_test
```

The Simulator used for asm generation is VCS `R-2020.12_Full64` and the ISS was Spike `1.0.1-dev`
@sammy17 please update this thread if you find a fix 🙂. Thanks