chipsalliance/riscv-dv #807: The "relocation truncated to fit" error occurs on a compilation of a code generated for the "ml" target.

dariuszst, 2021-07-12:
After increasing a number of instructions for ml target in the commit 2b2abda6fa0fdd35435cca615861fe3145996b14 the successful generation of a assembly code is only possible when  generation timeout is increased by "--gen_timeout". 
While it is not a big problem, but when code is finally generated in most cases it cannot be compiled with gcc because of a address relocation problem:
```
Mon, 12 Jul 2021 08:19:35 INFO     Creating output directory: sim_out_ml
Mon, 12 Jul 2021 08:19:35 INFO     Processing regression test list : .../riscv-dv/target/ml/testlist.yaml, test: all
Mon, 12 Jul 2021 08:19:35 INFO     Found matched tests: riscv_rand_test, iterations:1
Mon, 12 Jul 2021 08:19:35 INFO     Processing simulator setup file : .../riscv-dv/yaml/simulator.yaml
Mon, 12 Jul 2021 08:19:35 INFO     Found matching simulator: riviera
Mon, 12 Jul 2021 08:19:35 INFO     Building RISC-V instruction generator
Mon, 12 Jul 2021 08:20:11 INFO     Running RISC-V instruction generator
Mon, 12 Jul 2021 08:20:11 INFO     Generating 1 riscv_rand_test
Mon, 12 Jul 2021 08:20:11 INFO     Running riscv_rand_test with 1 batches
Mon, 12 Jul 2021 08:20:11 INFO     Running riscv_rand_test, batch 1/1, test_cnt:1
Mon, 12 Jul 2021 08:41:30 INFO     Compiling sim_out_ml/asm_test/riscv_rand_test_0.S
/tmp/cc4iIFSO.o: in function `sub_2_j6':
(.text+0x1e9234): relocation truncated to fit: R_RISCV_JAL against `sub_4'
```

The compilation command looks in this case something like:
`'riscv64-unknown-elf-gcc', '-static', '-mcmodel=medany', '-fvisibility=hidden', '-nostdlib', '-nostartfiles', 'sim_out_ml/asm_test/riscv_rand_test_0.S', '-I/.../riscv-dv/user_extension', '-T/.../riscv-dv/scripts/link.ld', '-o', 'sim_out_ml/asm_test/riscv_rand_test_0.o', '-mno-strict-align', '-march=rv64imc', '-mabi=lp64'`

mukesh891, 2022-08-16:
> After increasing a number of instructions for ml target in the commit [2b2abda](https://github.com/google/riscv-dv/commit/2b2abda6fa0fdd35435cca615861fe3145996b14) the successful generation of a assembly code is only possible when generation timeout is increased by "--gen_timeout". While it is not a big problem, but when code is finally generated in most cases it cannot be compiled with gcc because of a address relocation problem:
> 
> ```
> Mon, 12 Jul 2021 08:19:35 INFO     Creating output directory: sim_out_ml
> Mon, 12 Jul 2021 08:19:35 INFO     Processing regression test list : .../riscv-dv/target/ml/testlist.yaml, test: all
> Mon, 12 Jul 2021 08:19:35 INFO     Found matched tests: riscv_rand_test, iterations:1
> Mon, 12 Jul 2021 08:19:35 INFO     Processing simulator setup file : .../riscv-dv/yaml/simulator.yaml
> Mon, 12 Jul 2021 08:19:35 INFO     Found matching simulator: riviera
> Mon, 12 Jul 2021 08:19:35 INFO     Building RISC-V instruction generator
> Mon, 12 Jul 2021 08:20:11 INFO     Running RISC-V instruction generator
> Mon, 12 Jul 2021 08:20:11 INFO     Generating 1 riscv_rand_test
> Mon, 12 Jul 2021 08:20:11 INFO     Running riscv_rand_test with 1 batches
> Mon, 12 Jul 2021 08:20:11 INFO     Running riscv_rand_test, batch 1/1, test_cnt:1
> Mon, 12 Jul 2021 08:41:30 INFO     Compiling sim_out_ml/asm_test/riscv_rand_test_0.S
> /tmp/cc4iIFSO.o: in function `sub_2_j6':
> (.text+0x1e9234): relocation truncated to fit: R_RISCV_JAL against `sub_4'
> ```
> 
> The compilation command looks in this case something like: `'riscv64-unknown-elf-gcc', '-static', '-mcmodel=medany', '-fvisibility=hidden', '-nostdlib', '-nostartfiles', 'sim_out_ml/asm_test/riscv_rand_test_0.S', '-I/.../riscv-dv/user_extension', '-T/.../riscv-dv/scripts/link.ld', '-o', 'sim_out_ml/asm_test/riscv_rand_test_0.o', '-mno-strict-align', '-march=rv64imc', '-mabi=lp64'`

Hello @dariuszst ,
       Can you please try to change the "-mcmodel=medany" to "-mcmodel=medlow" in the run.py file and again try to compile and run, please let me know if its working.

Thanks

liangmin8743, 2023-07-05:
yes, same problem.... i have tried to change the "-mcmodel=medany" to "-mcmodel=medlow", but still the same

liangmin8743, 2023-07-05:
@mukesh891