chipsalliance/riscv-dv #816: support for bitmanip 1.0.0 in the near term

strichmo, 2021-08-25:
I am leading the Verification Task Group at OpenHW Group and one of the cores we have under development is implementing the frozen bitmanip extensions Zba, Zbb, Zbc, and Zbs.  It appears that riscv-dv's current B extension support is at the 0.92 draft version.  Is there a plan to update the instruction definitions and related constraints for any later versions of bitmanip in the near future.

mcockrell-google, 2021-08-25:
Yes, we are planning to add support for all instructions as the spec's evolve. Our current focus is the Vector extension. We are always looking for contributors if you're interested!

vogelpi, 2021-11-15:
Hi,
we will upgrade the bitmanip implementation in Ibex to v.1.0.0 and draft v.0.94 (for the sub-extensions that didn't make it into v.1.0.0). As part of that we also plan to re-align support in RISCV-DV. See https://github.com/lowRISC/ibex/issues/1470

I have a local copy of RISCV-DV with Zba working and hope to do a PR soonish.

silabs-hfegran, 2021-11-16:
Hi, 
I have added support for the v.1.0.0 extensions with PR #832 
Sub-extensions for v.0.9x have not been updated in this PR.

vogelpi, 2021-12-02:
For the remaining instructions not part of v.1.0.0, I have filed a PR here #834 to upgrade them to the latest draft.

vogelpi, 2021-12-03:
I am closing this issue as all related PR have been merged. RISCV-DV now supports bitmanip v.1.00 as well as the latest draft v.0.93 for the other sub-extensions that haven't been ratified yet.

weicaiyang, 2021-12-03:
> @weicaiyang I think this issue can be closed as all related PR have been merged. RISCV-DV now supports bitmanip v.1.00 as well as the latest draft v.0.93 for the other sub-extensions that haven't been ratified yet.

Thanks @vogelpi. Closing this

closed 2021-12-03 by weicaiyang