chipsalliance/riscv-dv #817: Error for pyboolector while install

nicolast0604, 2021-08-25:
Collecting pyboolector>=3.2.2 (from pyvsc->-r requirements.txt (line 10))
  Using cached https://files.pythonhosted.org/packages/9d/d2/4b244e2d3100b1d7af0e68b0181538d33d4a84e75305e296080eb74f26f2/PyBoolector-3.2.2.20210617.2.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-3ji4oktn/pyboolector/setup.py", line 14, in <module>
        with open(cmakelists_txt, "r") as f:
    FileNotFoundError: [Errno 2] No such file or directory: '/tmp/pip-build-3ji4oktn/CMakeLists.txt'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-3ji4oktn/pyboolector/
nioclast0604@nioclast0604-VirtualBox:~/riscv-dv$

andrewvaughanj, 2021-08-25:
Also reported here: https://github.com/Boolector/boolector/issues/176

Probably makes sense to track this against Boolector for now.

@nicolast0604 it would be useful to report your Python version and operating system details here to help the riscv-dv team understand your setup.

nicolast0604, 2021-08-25:
Ubuntu 18.04.5
Python 3.6.9

mballance, 2021-08-25:
In my experience, this is most-frequently caused by an older pip version. pyboolector needs to be installed as a binary wheel, and this requires an updated pip version.
Please try updating pip and re-installing pyboolector:
```
% python3 -m pip install --upgrade pip
% python3 -m pip install pyboolector
```

andrewvaughanj, 2021-08-25:
@mballance this was indeed the issue (although I wish I'd read your reply before trying to work this out myself :D)

closed 2021-09-17 by hodjat91