chipsalliance/riscv-dv #824: Synopsys VCS support for Google DV in ubuntu

sundas56, 2021-09-22:
Hello, I'm trying to install google DV in Red Hat, because I only have Red Hat version of  Sysnopsys VCS. But facing issues in installing git, there are some mirror issues.

hodjat91, 2021-09-22:
Hello,

Can you please share the specific error you get with us?

sundas56, 2021-09-22:
when I try to install git "yum install git", it shows this error.
"This system is not registered with an entitlement server. You can use subscription-manager to register".
Then on registering through  subscription-manager, I get this
"Unable to verify server's identity: certificate verify failed"

sundas56, 2021-09-22:
is it possible to run google RISC-V DV without simulator like vcs or questa. Is it possible to just copy the output files of google RISC-V DV process  to windows and run those files in simulator in windows.

closed 2021-11-04 by sundas56