chipsalliance/riscv-dv #830: VCS Error

sundas56, 2021-11-04:
on executing  both these commands
run --test=riscv_arithmetic_basic_test
run --test riscv_arithmetic_basic_test --iss spike
I got this error 
/bin/bash: line 0: exec: vcs: not found
Wed, 03 Nov 2021 21:51:27 ERROR    ERROR return code: True/127, cmd:vcs -file /home/ubuntu/riscv-dv/vcs.compile.option.f

hodjat91, 2021-11-04:
Do you have VCS license and have you set up the environment path? What do you see with running this command? 
"echo $VCS_HOME"

sundas56, 2021-11-04:
no I don't have VCS license but installed spike

haroonshafique-10xe, 2021-11-04:
Spike is an **Instruction Set** Simulator (ISS) which can run the generated assembly tests and not the SystemVerilog/UVM code. If you want to generate assembly tests from the riscv-dv, then you need either a simulator which supports UVM (e.g vcs, questa etc) or you need Python to be installed on your PC and use `pyflow`.

sundas56, 2021-11-04:
thanks, I'll try pyflow

sundas56, 2021-11-15:
what do you mean by "spike can only run generated assembly tests and not the SystemVerilog/UVM code". because riscv_arithmetic_basic_test  is an assembly code not the SV/UVM code, then why it's giving that error. Can you please explain.

sundas56, 2021-11-16:
> Spike is an **Instruction Set** Simulator (ISS) which can run the generated assembly tests and not the SystemVerilog/UVM code. If you want to generate assembly tests from the riscv-dv, then you need [...]

what do you mean by "spike can only run generated assembly tests and not the SystemVerilog/UVM code". because riscv_arithmetic_basic_test is an assembly code not the SV/UVM code, then why it's giving that error. Can you please explain.

haroonshafique-10xe, 2021-11-19:
@sundas56 , which error did it give while running an assembly test on spike? Can you also paste the command here for running the test on spike?

sundas56, 2021-11-20:
command
run --test riscv_arithmetic_basic_test --iss spike
error 
" /bin/bash: line 0: exec: vcs: not found
ERROR return code: True/127, cmd:vcs -file /home/ubuntu/riscv-dv/vcs.compile.option.f  "

tuppi-ovh, 2022-07-08:
@sundas56 by default the `vcs` simulator is used. You should specify explicitely a simulator you have using `--simulator` argument. For 100% open source solution try with `--simulator=pyflow`.