chipsalliance/riscv-dv #846: Page Table Exception Error and Fix

buraktoker, 2022-02-04:
Creating riscv_mmu_stress_test with this environment may cause a page table exception in spike ISS simulator. I faced the error when mret instruction committed in the end of init_user_mode function.
You can see the spike log in the below image.
[image]

I checked the problematic address in spike and I saw "0x200074f1" content cause problem. The reason behind is that non-leaf page table entries can not have D,A and U bits.  This information is given in Volume II: RISC-V Privileged Architectures V20211203:

"For non-leaf PTEs, the D, A, and U bits are reserved for future standard use. Until their use is defined by a standard extension, they must be cleared by software for forward compatibility."

To solve this problem, I have added  these extra lines to riscv_page_table_list.sv .
"
Line 189: valid_link_pte.a = 1'b0;
Line 190: valid_link_pte.d = 1'b0;
Line 191: valid_link_pte.u = 1'b0;
"

bieganski, 2022-02-24:
I confirm the issue, same holds with `riscv_u_mode_rand_test`.
In Spike, the check is performed in `riscv/mmu.cc`:

```c
} else if (PTE_TABLE(pte)) { // is not leaf
if (pte & (PTE_D | PTE_A | PTE_U | PTE_N | PTE_PBMT))
          break;
...
}
```

davine47, 2023-03-31:
It will make spike fall into an infinite looping.