chipsalliance/riscv-dv #861: Failed to generate test using pyflow.

lotharelvin, 2022-05-05:
Hi,
I have installed pyvsc but when I type 
`python3` run.py --target rv32i --simulator=pyflow
in the riscv-dv folder, it raised an exception and says Test generation failed.

[image]

Can anyone tell me how to fix this problem?
Thanks!

hodjat91, 2022-05-05:
Could it be because you haven't passed the test name? Can you pass this and try again? 
```
--test=riscv_arithmetic_basic_test
```

lotharelvin, 2022-05-05:
Yes, I forgot to specify the test name, after passing the test name it works. 
Thanks!

But I have another question, in the document it requires riscv-gcc as follows:
`export RISCV_TOOLCHAIN=<riscv_gcc_install_path>
export RISCV_GCC="$RISCV_TOOLCHAIN/bin/riscv32-unknown-elf-gcc"
export RISCV_OBJCOPY="$RISCV_TOOLCHAIN/bin/riscv32-unknown-elf-objcopy"
`
I configure the riscv-gcc with `--enable-multilib` but there is no riscv32-unknown-elf- in the toolchain/bin folder, should I configure the riscv-gcc and specifying a 32bit building environment like `--with-arch=rv32gc --with-abi=ilp32d` ?

Thanks!