chipsalliance/riscv-dv #869: Solve failure in vsc/model/randomizer.py during `riscv_rand_instr_test` test

tuppi-ovh, 2022-06-28:
The command to execute: `python3 run.py --test=riscv_rand_instr_test --iss=spike --simulator=pyflow`

The traceback of the error is below:

```
Tue, 28 Jun 2022 13:31:27 lib.py:126   INFO     Traceback (most recent call last):
  File "<path>/riscv-dv/pygen/pygen_src/test/riscv_instr_base_test.py", line 44, in run_phase
    self._run_phase(num)
  File "<path>/riscv-dv/pygen/pygen_src/test/riscv_instr_base_test.py", line 71, in _run_phase
    self.asm.gen_program()
  File "pygen/pygen_src/riscv_asm_program_gen.py", line 96, in gen_program
    self.gen_sub_program(hart, self.sub_program[hart],
  File "pygen/pygen_src/riscv_asm_program_gen.py", line 218, in gen_sub_program
    self.generate_directed_instr_stream(hart=hart,
  File "pygen/pygen_src/riscv_asm_program_gen.py", line 1175, in generate_directed_instr_stream
    new_instr_stream.randomize()
  File "<path>/riscv-dv/.venv/lib/python3.9/site-packages/vsc/rand_obj.py", line 171, in randomize
    raise e
  File "<path>/riscv-dv/.venv/lib/python3.9/site-packages/vsc/rand_obj.py", line 162, in randomize
    Randomizer.do_randomize(
  File "<path>/riscv-dv/.venv/lib/python3.9/site-packages/vsc/model/randomizer.py", line 607, in do_randomize
    fm.post_randomize()
  File "<path>/riscv-dv/.venv/lib/python3.9/site-packages/vsc/model/field_composite_model.py", line 148, in post_randomize
    self.rand_if.do_post_randomize()
  File "<path>/riscv-dv/.venv/lib/python3.9/site-packages/vsc/rand_obj.py", line 291, in do_post_randomize
    self.post_randomize()
  File "pygen/pygen_src/riscv_load_store_instr_lib.py", line 117, in post_randomize
    self.add_mixed_instr(self.num_mixed_instr)
  File "pygen/pygen_src/riscv_directed_instr_lib.py", line 87, in add_mixed_instr
    instr = self.randomize_instr(instr)
  File "pygen/pygen_src/riscv_instr_stream.py", line 238, in randomize_instr
    instr = self.randomize_gpr(instr)
  File "pygen/pygen_src/riscv_instr_stream.py", line 260, in randomize_gpr
    instr.rs1 != cfg.reserved_regs[i]
  File "<path>/riscv-dv/.venv/lib/python3.9/site-packages/vsc/rand_obj.py", line 271, in __exit__
    raise e
  File "<path>/riscv-dv/.venv/lib/python3.9/site-packages/vsc/rand_obj.py", line 261, in __exit__
    Randomizer.do_randomize(
  File "<path>/riscv-dv/.venv/lib/python3.9/site-packages/vsc/model/randomizer.py", line 597, in do_randomize
    r.randomize(ri, bounds_v.bound_m)
  File "<path>/riscv-dv/.venv/lib/python3.9/site-packages/vsc/model/randomizer.py", line 215, in randomize
    raise SolveFailure(
vsc.model.solve_failure.SolveFailure: solve failure
Traceback (most recent call last):
  File "<path>/riscv-dv/pygen/pygen_src/test/riscv_instr_base_test.py", line 44, in run_phase
    self._run_phase(num)
  File "<path>/riscv-dv/pygen/pygen_src/test/riscv_instr_base_test.py", line 71, in _run_phase
    self.asm.gen_program()
  File "pygen/pygen_src/riscv_asm_program_gen.py", line 96, in gen_program
    self.gen_sub_program(hart, self.sub_program[hart],
  File "pygen/pygen_src/riscv_asm_program_gen.py", line 218, in gen_sub_program
    self.generate_directed_instr_stream(hart=hart,
  File "pygen/pygen_src/riscv_asm_program_gen.py", line 1175, in generate_directed_instr_stream
    new_instr_stream.randomize()
  File "<path>/riscv-dv/.venv/lib/python3.9/site-packages/vsc/rand_obj.py", line 171, in randomize
    raise e
  File "<path>/riscv-dv/.venv/lib/python3.9/site-packages/vsc/rand_obj.py", line 162, in randomize
    Randomizer.do_randomize(
  File "<path>/riscv-dv/.venv/lib/python3.9/site-packages/vsc/model/randomizer.py", line 607, in do_randomize
    fm.post_randomize()
  File "<path>/riscv-dv/.venv/lib/python3.9/site-packages/vsc/model/field_composite_model.py", line 148, in post_randomize
    self.rand_if.do_post_randomize()
  File "<path>/riscv-dv/.venv/lib/python3.9/site-packages/vsc/rand_obj.py", line 291, in do_post_randomize
    self.post_randomize()
  File "pygen/pygen_src/riscv_load_store_instr_lib.py", line 117, in post_randomize
    self.add_mixed_instr(self.num_mixed_instr)
  File "pygen/pygen_src/riscv_directed_instr_lib.py", line 87, in add_mixed_instr
    instr = self.randomize_instr(instr)
  File "pygen/pygen_src/riscv_instr_stream.py", line 238, in randomize_instr
    instr = self.randomize_gpr(instr)
  File "pygen/pygen_src/riscv_instr_stream.py", line 260, in randomize_gpr
    instr.rs1 != cfg.reserved_regs[i]
  File "<path>/riscv-dv/.venv/lib/python3.9/site-packages/vsc/rand_obj.py", line 271, in __exit__
    raise e
  File "<path>/riscv-dv/.venv/lib/python3.9/site-packages/vsc/rand_obj.py", line 261, in __exit__
    Randomizer.do_randomize(
  File "<path>/riscv-dv/.venv/lib/python3.9/site-packages/vsc/model/randomizer.py", line 597, in do_randomize
    r.randomize(ri, bounds_v.bound_m)
  File "<path>/riscv-dv/.venv/lib/python3.9/site-packages/vsc/model/randomizer.py", line 215, in randomize
    raise SolveFailure(
vsc.model.solve_failure.SolveFailure: solve failure
Solve failure: set 'solve_fail_debug=1' for more details
Solve failure: set 'solve_fail_debug=1' for more details
Solve failure: set 'solve_fail_debug=1' for more details
Solve failure: set 'solve_fail_debug=1' for more details
Traceback (most recent call last):
  File "<path>/riscv-dv/pygen/pygen_src/test/riscv_instr_base_test.py", line 88, in <module>
    riscv_base_test_ins.run()
  File "<path>/riscv-dv/pygen/pygen_src/test/riscv_instr_base_test.py", line 40, in run
    raise Exception("Test-generation jobs failed")
Exception: Test-generation jobs failed
```

Maybe the error can be "solved" by fixing a correct versions in `requirements.txt`?

DivyaK28, 2022-07-13:
Same here
```
python run.py --target rv64imafdc --iss spike --simulator pyflow --testlist $PWD//target/rv64imafdc/testlist.yaml
Wed, 13 Jul 2022 17:54:14 INFO     Creating output directory: out_2022-07-13
Wed, 13 Jul 2022 17:54:14 INFO     Processing regression test list : /home/divya/riscv-dv//target/rv64imafdc/testlist.yaml, test: all
Wed, 13 Jul 2022 17:54:14 INFO     Processing regression test list : /home/divya/riscv-dv/target/rv64imc/testlist.yaml, test: all
Wed, 13 Jul 2022 17:54:14 INFO     Processing regression test list : /home/divya/riscv-dv/target/rv32imc/testlist.yaml, test: all
Wed, 13 Jul 2022 17:54:14 INFO     Processing regression test list : /home/divya/riscv-dv/yaml/base_testlist.yaml, test: all
Wed, 13 Jul 2022 17:54:14 INFO     Found matched tests: riscv_arithmetic_basic_test, iterations:2
Wed, 13 Jul 2022 17:54:14 INFO     Found matched tests: riscv_rand_instr_test, iterations:2
Wed, 13 Jul 2022 17:54:14 INFO     Found matched tests: riscv_jump_stress_test, iterations:2
Wed, 13 Jul 2022 17:54:14 INFO     Found matched tests: riscv_loop_test, iterations:2
Wed, 13 Jul 2022 17:54:14 INFO     Found matched tests: riscv_rand_jump_test, iterations:2
Wed, 13 Jul 2022 17:54:14 INFO     Found matched tests: riscv_mmu_stress_test, iterations:2
Wed, 13 Jul 2022 17:54:14 INFO     Found matched tests: riscv_no_fence_test, iterations:2
Wed, 13 Jul 2022 17:54:14 INFO     Found matched tests: riscv_illegal_instr_test, iterations:2
Wed, 13 Jul 2022 17:54:14 INFO     Found matched tests: riscv_ebreak_test, iterations:2
Wed, 13 Jul 2022 17:54:14 INFO     Found matched tests: riscv_ebreak_debug_mode_test, iterations:2
Wed, 13 Jul 2022 17:54:14 INFO     Found matched tests: riscv_full_interrupt_test, iterations:2
Wed, 13 Jul 2022 17:54:14 INFO     Found matched tests: riscv_unaligned_load_store_test, iterations:1
Wed, 13 Jul 2022 17:54:14 INFO     Found matched tests: riscv_non_compressed_instr_test, iterations:1
Wed, 13 Jul 2022 17:54:14 INFO     Found matched tests: riscv_hint_instr_test, iterations:2
Wed, 13 Jul 2022 17:54:14 INFO     Found matched tests: riscv_pmp_test, iterations:2
Wed, 13 Jul 2022 17:54:14 INFO     Found matched tests: riscv_machine_mode_rand_test, iterations:2
Wed, 13 Jul 2022 17:54:14 INFO     Found matched tests: riscv_privileged_mode_rand_test, iterations:2
Wed, 13 Jul 2022 17:54:14 INFO     Found matched tests: riscv_invalid_csr_test, iterations:2
Wed, 13 Jul 2022 17:54:14 INFO     Found matched tests: riscv_sfence_exception_test, iterations:2
Wed, 13 Jul 2022 17:54:14 INFO     Found matched tests: riscv_amo_test, iterations:2
Wed, 13 Jul 2022 17:54:14 INFO     Found matched tests: riscv_floating_point_arithmetic_test, iterations:1
Wed, 13 Jul 2022 17:54:14 INFO     Found matched tests: riscv_floating_point_rand_test, iterations:1
Wed, 13 Jul 2022 17:54:14 INFO     Found matched tests: riscv_floating_point_mmu_stress_test, iterations:2
Wed, 13 Jul 2022 17:54:14 INFO     Processing simulator setup file : /home/divya/riscv-dv/yaml/simulator.yaml
Wed, 13 Jul 2022 17:54:14 INFO     Found matching simulator: pyflow
Wed, 13 Jul 2022 17:54:14 INFO     Building RISC-V instruction generator
Wed, 13 Jul 2022 17:54:14 INFO     Running RISC-V instruction generator
Wed, 13 Jul 2022 17:54:14 INFO     Generating 2 riscv_arithmetic_basic_test
Wed, 13 Jul 2022 17:54:14 INFO     Running riscv_arithmetic_basic_test with 1 batches
Wed, 13 Jul 2022 17:54:14 INFO     Running riscv_arithmetic_basic_test, batch 1/1, test_cnt:2
Wed, 13 Jul 2022 17:55:37 INFO     Generating 2 riscv_rand_instr_test
Wed, 13 Jul 2022 17:55:37 INFO     Running riscv_rand_instr_test with 1 batches
Wed, 13 Jul 2022 17:55:37 INFO     Running riscv_rand_instr_test, batch 1/1, test_cnt:2
Wed, 13 Jul 2022 18:15:37 ERROR    Timeout[1200s]:  python3 /home/divya/riscv-dv/pygen/pygen_src/test/riscv_instr_base_test.py --num_of_tests=2 --start_idx=0 --asm_file_name=out_2022-07-13/asm_test/riscv_rand_instr_test --log_file_name=out_2022-07-13/sim_riscv_rand_instr_test_0.log  --target=rv64imafdc  --gen_test=riscv_instr_base_test  --seed=1570394483 --instr_cnt=10000 --num_of_sub_program=5 --directed_instr_0=riscv_load_store_rand_instr_stream,4 --directed_instr_1=riscv_loop_instr,4 --directed_instr_2=riscv_hazard_instr_stream,4 --directed_instr_3=riscv_load_store_hazard_instr_stream,4 --directed_instr_4=riscv_multi_page_load_store_instr_stream,4 --directed_instr_5=riscv_mem_region_stress_test,4 --directed_instr_6=riscv_jal_instr,4

Wed, 13 Jul 2022 18:15:37 INFO     Generating 2 riscv_jump_stress_test
Wed, 13 Jul 2022 18:15:37 INFO     Running riscv_jump_stress_test with 1 batches
Wed, 13 Jul 2022 18:15:37 INFO     Running riscv_jump_stress_test, batch 1/1, test_cnt:2
Wed, 13 Jul 2022 18:20:22 INFO     Traceback (most recent call last):
  File "/home/divya/riscv-dv/pygen/pygen_src/test/riscv_instr_base_test.py", line 44, in run_phase
    self._run_phase(num)
  File "/home/divya/riscv-dv/pygen/pygen_src/test/riscv_instr_base_test.py", line 71, in _run_phase
    self.asm.gen_program()
  File "pygen/pygen_src/riscv_asm_program_gen.py", line 115, in gen_program
    sub_program_name, cfg.num_of_sub_program)
  File "pygen/pygen_src/riscv_asm_program_gen.py", line 232, in gen_callstack
    self.callstack_gen.init(num_sub_program + 1)
  File "/home/divya/.pyenv/versions/py37/lib/python3.7/site-packages/vsc/rand_obj.py", line 90, in __getattribute__
    ret = object.__getattribute__(self, a)
[... cut ...]
    raise Exception("Test-generation jobs failed")
Exception: Test-generation jobs failed

Wed, 13 Jul 2022 18:20:22 ERROR    ERROR return code: True/1, cmd: python3 /home/divya/riscv-dv/pygen/pygen_src/test/riscv_instr_base_test.py --num_of_tests=2 --start_idx=0 --asm_file_name=out_2022-07-13/asm_test/riscv_jump_stress_test --log_file_name=out_2022-07-13/sim_riscv_jump_stress_test_0.log  --target=rv64imafdc  --gen_test=riscv_instr_base_test  --seed=1367358632 --instr_cnt=5000 --num_of_sub_program=5 --directed_instr_1=riscv_jal_instr,20

```

fPecc, 2022-07-21:
Same happened to me.

mczyz-antmicro, 2023-05-12:
I've managed to reproduce the issue and am looking into possible fixes

AroundKiki, 2024-01-02:
same problem

Xxfore, 2024-03-13:
sample problem still exist on master branch.

CVYang, 2024-11-02:
same problem still unfixed

AllenZF, 2025-12-17:
Any progress on this issue?