chipsalliance/riscv-dv #879: Enabling BitManip instructions generation

algrobman, 2022-07-26:
How to enable bitmanip instructions to be generated?

I've added RV32B to  supported_isa array in riscv_core_setting.sv file with no success ..

silabs-hfegran, 2022-07-28:
Are you intending to use the old 0.92/3  or the 1.0.0 version of the bitmanip instructions? 
If you want to generate for the new spec, you will need to set/constrain the cfg.enable_zb*_extension-fields.
Similarly, for the old spec, you need cfg.enable_b_extension and the respective cfg.enable_bitmanip_groups (ZB*)

algrobman, 2022-07-28:
I've added all RV32ZB* and RV32B to the list of supported ISA, but  no bitmanip instructions were generated ...

qaziullah, 2022-12-20:
Hi @algrobman I was having the same issue. For me issue was resolved by 

1.  adding +enable_zb*_extension option in the test.yaml file present in rv32imcb target folder.
2.  adding RV32ZB* in the supported ISA list ( and removing RV32B)

silabs-hfegran, 2022-12-21:
> Hi @algrobman I was having the same issue. For me issue was resolved by
> 
> 1. adding +enable_zb*_extension option in the test.yaml file present in rv32imcb target folder.
> 2. adding RV32ZB* in the supported ISA list ( and removing RV32B)

Thanks to @qaziullah for spelling out the details, this is indeed the intended way of configuring these extensions