chipsalliance/riscv-dv #898: python generator support?

InspireSemi, 2022-09-15:
I see the pygen dir and can generate arithmetic test code.  

But the testlist.yaml files seem to be missing from the target/ dir under pygen.  So it is unable to generate some of the tests that the Readme.md says it supports. 

Is this just a matter of copying the .yaml files from the top level target/ dir to the one under pygen?

Also what is the current status of pygen?  Cann we use it to generate but not run test cases?

aneels3, 2022-11-13:
Hi @InspireSemi 
Pygen is integrated within riscv-dv flow in such a way that it doesn't require to have testlist.yaml in target/ directory.
You can check the [readme.md](https://github.com/google/riscv-dv/tree/master/pygen/pygen_src#readme) of pygen to run the supported test and targets.
No need to copy any testlist file under pygen directory to run the supported tests.

Best,
Anil

InspireSemi, 2022-11-14:
I can generate the riscv_arithmetic_basic_test, using the --simulator=pyflow option.

If I try to generate all of the tests : python3 run.py --simulator=pyflow --steps gen --target rv64imafdc
It only generates the basic_arithmetic tests:  

Mon, 14 Nov 2022 11:27:33 INFO     Found matching simulator: pyflow
Mon, 14 Nov 2022 11:27:33 INFO     Building RISC-V instruction generator
Mon, 14 Nov 2022 11:27:33 INFO     Running RISC-V instruction generator
Mon, 14 Nov 2022 11:27:33 INFO     Generating 2 riscv_arithmetic_basic_test
Mon, 14 Nov 2022 11:27:33 INFO     Running riscv_arithmetic_basic_test with 1 batches
Mon, 14 Nov 2022 11:27:33 INFO     Running riscv_arithmetic_basic_test, batch 1/1, test_cnt:2
Mon, 14 Nov 2022 11:31:41 INFO     Generating 2 riscv_rand_instr_test
Mon, 14 Nov 2022 11:31:41 INFO     Running riscv_rand_instr_test with 1 batches
Mon, 14 Nov 2022 11:31:41 INFO     Running riscv_rand_instr_test, batch 1/1, test_cnt:2
Mon, 14 Nov 2022 11:51:41 ERROR    Timeout[1200s]:  python3 /home/mkarasek/src/riscv-dv/pygen/pygen_src/test/riscv_instr_base_test.py --num_of_tests=2 --start_idx=0 --asm_file_name=out_2022-11-14/asm_test/riscv_rand_instr_test --log_file_name=out_2022-11-14/sim_riscv_rand_instr_test_0.log  --target=rv64imafdc  --gen_test=riscv_instr_base_test  --seed=923921243 --instr_cnt=10000 --num_of_sub_program=5 --directed_instr_0=riscv_load_store_rand_instr_stream,4 --directed_instr_1=riscv_loop_instr,4 --directed_instr_2=riscv_hazard_instr_stream,4 --directed_instr_3=riscv_load_store_hazard_instr_stream,4 --directed_instr_4=riscv_multi_page_load_store_instr_stream,4 --directed_instr_5=riscv_mem_region_stress_test,4 --directed_instr_6=riscv_jal_instr,4

Mon, 14 Nov 2022 11:51:41 INFO     Generating 2 riscv_jump_stress_test
Mon, 14 Nov 2022 11:51:41 INFO     Running riscv_jump_stress_test with 1 batches
Mon, 14 Nov 2022 11:51:41 INFO     Running riscv_jump_stress_test, batch 1/1, test_cnt:2
Mon, 14 Nov 2022 12:06:03 INFO     Traceback (most recent call last):
  File "/home/mkarasek/src/riscv-dv/pygen/pygen_src/test/riscv_instr_base_test.py", line 44, in run_phase
    self._run_phase(num)
  File "/home/mkarasek/src/riscv-dv/pygen/pygen_src/test/riscv_instr_base_test.py", line 71, in _run_phase
    self.asm.gen_program()
  File "pygen/pygen_src/riscv_asm_program_gen.py", line 115, in gen_program
    sub_program_name, cfg.num_of_sub_program)
  File "pygen/pygen_src/riscv_asm_program_gen.py", line 232, in gen_callstack
    self.callstack_gen.init(num_sub_program + 1)
  File "/home/mkarasek/.local/lib/python3.6/site-packages/vsc/rand_obj.py", line 90, in __getattribute__
    ret = object.__getattribute__(self, a)
AttributeError: 'riscv_asm_program_gen' object has no attribute 'callstack_gen'
Traceback (most recent call last):
  File "/home/mkarasek/src/riscv-dv/pygen/pygen_src/test/riscv_instr_base_test.py", line 44, in run_phase
    self._run_phase(num)
  File "/home/mkarasek/src/riscv-dv/pygen/pygen_src/test/riscv_instr_base_test.py", line 71, in _run_phase
    self.asm.gen_program()
  File "pygen/pygen_src/riscv_asm_program_gen.py", line 115, in gen_program
    sub_program_name, cfg.num_of_sub_program)
  File "pygen/pygen_src/riscv_asm_program_gen.py", line 232, in gen_callstack
    self.callstack_gen.init(num_sub_program + 1)
  File "/home/mkarasek/.local/lib/python3.6/site-packages/vsc/rand_obj.py", line 90, in __getattribute__
    ret = object.__getattribute__(self, a)
AttributeError: 'riscv_asm_program_gen' object has no attribute 'callstack_gen'
Traceback (most recent call last):
  File "/home/mkarasek/src/riscv-dv/pygen/pygen_src/test/riscv_instr_base_test.py", line 88, in <module>
    riscv_base_test_ins.run()
  File "/home/mkarasek/src/riscv-dv/pygen/pygen_src/test/riscv_instr_base_test.py", line 40, in run
    raise Exception("Test-generation jobs failed")
Exception: Test-generation jobs failed

Mon, 14 Nov 2022 12:06:03 ERROR    ERROR return code: True/1, cmd: python3 /home/mkarasek/src/riscv-dv/pygen/pygen_src/test/riscv_instr_base_test.py --num_of_tests=2 --start_idx=0 --asm_file_name=out_2022-11-14/asm_test/riscv_jump_stress_test --log_file_name=out_2022-11-14/sim_riscv_jump_stress_test_0.log  --target=rv64imafdc  --gen_test=riscv_instr_base_test  --seed=1808718627 --instr_cnt=5000 --num_of_sub_program=5 --directed_instr_1=riscv_jal_instr,20

We moved back to using xcelium tools to gernerate everything and modifying the SV code to do pre/post init setup, etc..

mczyz-antmicro, 2023-05-12:
I've managed to reproduce the issue and am looking into possible fixes