chipsalliance/riscv-dv #927: kIllegalCompressedOpcode generates valid instruction (```c.fswsp```) when XLEN=32

atshinbo-etrinity, 2023-03-24:
**Overview**

The riscv-dv will generate ```kIllegalCompressedOpcode``` instruction sequence in the ```riscv_illegal_instr_test``` when the target supports C-extension.

```kIllegalCompressedOpcode``` have to generate undefined/invalid instruction code on RV32C/RV64C.
However, the problem is that valid instruction ```c.fswsp``` may included in ```kIllegalCompressedOpcode``` when target=rv32imafdc.

```c.fswsp``` is valid instruction in RV32C. 
Executing this instruction is not expected in the program context. So, storing data to the stack memory may cause a corruption of the program flow and may cause unexpected behavior.

It seems other compressed floating point instructions also should be excluded form ```kIllegalCompressedOpcode``` when C+F/D-extention is enabled.

**How to reproduce**

```
python3 ./run.py --target rv32imafdc -si vcs --isa rv32imafdc -m ilp32 -tn riscv_illegal_instr_test -s gen,gcc_compile -o out --start_seed 200 -i 10
```

- Generated code

```
out/asm_test/riscv_illegal_instr_test_1.S

                csrrsi       tp, 0x340, 27
                .4byte 0xe4ee # kIllegalCompressedOpcode
                mul          a0, a0, t6
```

- Disassemble
```
${OBJDUMP} -d --disassemble-all --disassemble-zeroes out/asm_test/riscv_illegal_instr_test_1.o

8000b984:       340de273                csrrsi  tp,mscratch,27
8000b988:       e4ee                    fsw     fs11,72(sp)        <--- valid instruction
8000b98a:       0000                    unimp
8000b98c:       03f50533                mul     a0,a0,t6
```

**Investigation**

- The riscv-dv defines ```legal_c10_opcode``` as following. ```3'b111``` is included only when XLEN=64.
```
riscv-dv/src/riscv_illegal_instr.sv

  bit [2:0]  legal_c10_opcode[$] = '{3'b000,
                                     3'b010,
                                     3'b100,
                                     3'b110};

  function void init(riscv_instr_gen_config cfg);
    if (riscv_instr_pkg::RV64I inside {riscv_instr_pkg::supported_isa}) begin
      legal_c00_opcode = {legal_c00_opcode, 3'b011, 3'b111};
      legal_c10_opcode = {legal_c10_opcode, 3'b011, 3'b111};
    end
```

- According to riscv-spec-20191213.pdf, ([15:13]=3'b111, [1:0]=2'b10) is also valid in RV32C as ```C.FSWSP```.
[image]

**Version**

```
commit b032c397bc2fad23bd87ff2bacd6e2dc731fd318
Merge: 4cf18db 8a6e482
Author: Pradheep Kaliraj <41653565+pradheepkaliraj@users.noreply.github.com>
Date:   Tue Feb 14 21:14:16 2023 -0800
```

franAyachi, 2025-02-05:
Hi! I am trying to reproduce this tests, but having  +illegal_instr_ratio=35 with a target that support C extension generates all the different illegal instruction unless the compress one....

atsushi-shinbo-esoltrinity, 2025-02-10:
```riscv_illegal_instr_test``` will genetate several illegal instruction codes, but it should not genetate valid instructions code in the target. The problem in this issue is ```c.fswsp``` instruction which is valid in the target rv32imafdc has been generated.

I confirmed that the problem is still occurred in latest master branch.
I believe the PR #921 fixed this problem but this PR has not yet been merged.

```
$ git clone https://github.com/chipsalliance/riscv-dv
$ cd riscv-dv
$ git branch
* master

$ git log -n 1
commit 7e54b678ab7499040336255550cdbd99ae887431
Merge: f0c570d fb6181b
Author: Tomasz Michalak <tmichalak@antmicro.com>
Date:   Fri Feb 7 10:50:06 2025 +0100

$ python3 ./run.py \
  --target rv32imafdc -si xlm --isa rv32imafdc -m ilp32 \
  -tn riscv_illegal_instr_test -s gen,gcc_compile \
  -o out --start_seed 8 -i 30 \
  --sim_opts "+enable_floating_point=1 +illegal_instr_ratio=35"

$ cat out/asm_test/riscv_illegal_instr_test_25.S
    ...
    .4byte 0x67b6 # kIllegalCompressedOpcode  -> valid instruction "flw fa5,76(sp)"
    ...
    .4byte 0xe4ee # kIllegalCompressedOpcode  -> valid instruction "fsw fs11,72(sp)"
    ...
```