chipsalliance/riscv-dv #934: pygen requirements installation failure

ElectronicBatman, 2023-05-15:
I am trying to use pygen and installing requirements as per documentation.
While executing following step : 
**pip3 install -r requirements.txt    # install dependencies (only once)**

getting following error:

Collecting pyboolector>=3.2.2 (from pyvsc->-r requirements.txt (line 10))
  Downloading https://files.pythonhosted.org/packages/82/be/1af424cf5dd10ae0e18478bea5b074ac1e7a13a3d46f43a6a4f341626f0c/PyBoolector-3.2.2.20230110.4.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-db4ojq0k/pyboolector/setup.py", line 14, in <module>
        with open(cmakelists_txt, "r") as f:
    FileNotFoundError: [Errno 2] No such file or directory: '/tmp/pip-build-db4ojq0k/CMakeLists.txt'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-db4ojq0k/pyboolector/
chipyard@6d203793f460:/root/my-chipyard/riscv-dv# python3
Python 3.6.9 (default, Jan 26 2021, 15:33:00)

hodjat91, 2023-05-15:
What OS are you running? Could you update your python version and retry? I've tested with python 3.9 and 3.10, both worked. Also, please take a look at this similar issue (https://github.com/fvutils/pyvsc/issues/54)

ElectronicBatman, 2023-05-16:
Python 3.6.9 (default, Jan 26 2021, 15:33:00) 
[GCC 8.4.0] on linux

zephray, 2023-05-25:
I am facing the same issue, my python version is
Python 3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0] on linux

Damien-Wu, 2023-07-19:
I  met the same issue with python version 3.8 ,and solved it with python 3.9.