chipsalliance/riscv-dv #937: Test cases generated for riscv64gc target do not run on latest version of spike

KatCe, 2023-05-28:
When generating the default set of regression tests with the following command:
```
run --target=rv64gc --simulator=questa -o out_rv64gc
```
and using the spike isa simulator cloned from the link (https://github.com/riscv-software-src/riscv-isa-sim, commit 740e6353a166ec3ba936b200482d6fc381b8082b) in the HTML documentation preview (https://htmlpreview.github.io/?https://github.com/google/riscv-dv/blob/master/docs/build/singlehtml/index.html#document-index), the tests fail in spike with the following error:

```
core   0: 0x000000008000b0f8 (0x000c00e7) jalr    s8
core   0: 3 0x000000008000b0f8 (0x000c00e7) x1  0x000000008000b0fc
core   0: >>>>  test_done
core   0: 0x000000008000a08a (0x00004185) c.li    gp, 1
core   0: 3 0x000000008000a08a (0x4185) x3  0x0000000000000001
core   0: 0x000000008000a08c (0x00000073) ecall
core   0: exception trap_machine_ecall, epc 0x000000008000a08c
terminate called after throwing an instance of 'std::runtime_error'
  what():  bad syscall #0
```
in out_rv64gc/spike_sim/riscv_arithmetic_basic_test.0.log

Test generation and spike simulation work fine for the default target rv32imc.

Test generation and spike simulation work fine for target rv64gc with spike cloned from https://github.com/lowRISC/riscv-isa-sim.git (commit 9af9730baf7b956c3072c1b436d867aca5ef8f4c).

ZJOJOw, 2023-06-08:
I have encountered the same problem, have you solved it?

KatCe, 2023-06-08:
No, not for the original spike.
The tests ran with with spike cloned from https://github.com/lowRISC/riscv-isa-sim.git (commit 9af9730baf7b956c3072c1b436d867aca5ef8f4c).

ZJOJOw, 2023-06-08:
which riscv_toolchain are you using? is it riscv32/64-unknown-elf-gcc?

liangmin8743, 2023-06-26:
you should pass --misaligned option to spike to enable misaligned memory accesses

KatCe, 2023-07-04:
The latest version of spike has no --misaligned argument and the build configuration option --enable-misaligned does not exist in the latest version of spike.

liangmin8743, 2023-07-05:
Check your iss simulation log to see if the exception "trap_store_address_misaligned" occured or not， if this is the case, it means that your spike does not support unaligned memory access. otherwise, you should try another solution

KatCe, 2023-07-18:
Yes, this exception occurs with the original spike, but not with the lowRISC fork. Would be good to mention in the documentation which version of Spike should be used.

riscv1111, 2023-10-11:
I'm also having the same issue.  Is it solved ?

zyx911126, 2023-10-23:
can you tell me the commit id of your riscv-dv？

omerguzelelectronicguy, 2025-02-04:
I don't have an unaligned accsess for load and store in my architecture. So I already wanted to go exception when there is unaligned load store. However, the termination is not done correctly. It just print bad syscall and exit after an `ecall` instruction. I thought it is related to this issue. Do you have any idea to solve this problem?