chipsalliance/riscv-dv #945: how to generate compressed instructions regression test

GMYMingyu, 2023-08-29:
Hi,
1）Objective：I would like to generate compressed instruction regression test with riscv-dv, which means the amount of compressed instrucions is large, exceeding 50% of all instructions
2) My action: I set a new test name in the yaml: +instr_cnt=1000 +disable_compressed_instr=0 +num_of_sub_program=0, accoding to the Section Runtime options of the generator on https://htmlpreview.github.io/?https://github.com/google/riscv-dv/blob/master/docs/build/singlehtml/index.html#runtime-options-of-the-generator. However, the generated .S file does not list any compressed instruction.
In my expectation, set +disable_compressed_instr=0 could generate multiple compressed instructions in the generated .s file, riscv-dv randomly generating compressed instructions.
Thanks

davine47, 2023-08-29:
hi，which generation flow you have used，python or systemverilog？

GMYMingyu, 2023-08-29:
Hi,
I am using the python. My generation flow is as follows:
1) setting the gen_opt in yaml
2) use the run.py . python36 run.py --isa rv64gcv --target rv64gcv -tn
compressed_instr -o /compressed_instr
Thanks for reply.

Mingyu

GMYMingyu, 2023-08-31:
> hi，which generation flow you have used，python or systemverilog？

Thanks for your hint.
I fixed this issue with the following step:
1) to generate compressed instruction regression test, configuring [march_isa] is also quite important in the yaml file. Add the RV32C to the isa gen_opt. Otherwise the riscv-dv would still disable the compressed_instr.
2) check the generation flow: Pyflow and SVflow have different flags for the gen_opt flag. svflow: +march=RV32C,RV32I,RV64I.... pyflow: +march_isa=RV32C,RV64C

closed 2023-08-31 by GMYMingyu