chipsalliance/riscv-dv #958: use ovpsim .Why is there no verbose log instruction

moyouth, 2023-10-07:
[image]
[image]
[image]

eroom1966, 2023-10-09:
try running with the flags --help or --helpall
there are a number of trace options, for example, I would recommend first selecting

--trace --tracechange

but there are many other trace options

moyouth, 2023-10-10:
[image]
ovpsim dont support --trace --tracechange，i add 2 commands in the picture,still no verbose log instruction

moyouth, 2023-10-10:
> try running with the flags --help or --helpall there are a number of trace options, for example, I would recommend first selecting
> 
> --trace --tracechange
> 
> but there are many other trace options
[image]

duncangraham-Imperas, 2023-10-10:
When you ask about _**verbose log instruction**_, what is it you are wanting to see?

There are two free products that include the OVP RISC-V reference models riscvOVPsim and riscvOVPsimPlus; both contain the same models but provide different features. You can obtain riscvOVPsimPlus from the [OVP World website](https://www.ovpworld.org/riscvOVPsimPlus/) and you should be able to switch to using this in the riscv-dv infrastructure; this will provide instruction trace output; if that s what you are looking for?

moyouth, 2023-10-10:
> When you ask about _**verbose log instruction**_, what is it you are wanting to see?
>
> There are two free products that include the OVP RISC-V reference models riscvOVPsim and riscvOVPsimPlus; both [...]
[image]

THANKS！！！！
using --iss=spike，the log in the picture like this：
In the log, the first column is the PC value, the second column is the instruction, the third column is the identifier for general registers or control registers, and the fourth column is the value of the register.

 i wanna see verbose log instruction when i use --iss=ovpsim. i dont use riscvOVPsimplus