chipsalliance/riscv-dv #964: Zicsr requirement for RV32I tests in pyflow

5hayanB, 2023-11-24:
I am using the following command to generate arithmetic tests for `rv32i` target

```shell
python3 run.py --target rv32i -o $HOME/temporary/riscv-dv -tn riscv_arithmetic_basic_test -i 10 -si pyflow --iss spike -v
```

However, I am receiving assembler errors at the compilation step

```shell
lib.py:198   DEBUG    $HOME/applications/riscv/toolchain/bin/riscv64-unknown-elf-gcc -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles $HOME/temporary/riscv-dv/asm_test/riscv_arithmetic_basic_test_0.S -I$HOME/repositories/other_repositories/github/riscv-dv/user_extension -T$HOME/repositories/other_repositories/github/riscv-dv/scripts/link.ld -o $HOME/temporary/riscv-dv/asm_test/riscv_arithmetic_basic_test_0.o -march=rv32i -mabi=ilp32
$HOME/temporary/riscv-dv/asm_test/riscv_arithmetic_basic_test_0.S: Assembler messages:
$HOME/temporary/riscv-dv/asm_test/riscv_arithmetic_basic_test_0.S:7: Error: unrecognized opcode `csrr x5,0xf14', extension `zicsr' required
$HOME/temporary/riscv-dv/asm_test/riscv_arithmetic_basic_test_0.S:15: Error: unrecognized opcode `csrw 0x301,x24', extension `zicsr' required
$HOME/temporary/riscv-dv/asm_test/riscv_arithmetic_basic_test_0.S:22: Error: unrecognized opcode `csrw 0x305,x24', extension `zicsr' required

Fri, 24 Nov 2023 12:48:42 lib.py:206   DEBUG    b''
Traceback (most recent call last):
  File "$HOME/repositories/other_repositories/github/riscv-dv/run.py", line 1170, in <module>
    main()
  File "$HOME/repositories/other_repositories/github/riscv-dv/run.py", line 1147, in main
    gcc_compile(matched_list, output_dir, args.isa, args.mabi,
  File "$HOME/repositories/other_repositories/github/riscv-dv/run.py", line 457, in gcc_compile
    run_cmd_output(cmd.split(), debug_cmd=debug_cmd)
  File "$HOME/repositories/other_repositories/github/riscv-dv/scripts/lib.py", line 207, in run_cmd_output
    raise exc
  File "$HOME/repositories/other_repositories/github/riscv-dv/scripts/lib.py", line 204, in run_cmd_output
    output = subprocess.check_output(cmd)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/subprocess.py", line 466, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['$HOME/applications/riscv/toolchain/bin/riscv64-unknown-elf-gcc', '-static', '-mcmodel=medany', '-fvisibility=hidden', '-nostdlib', '-nostartfiles', '$HOME/temporary/riscv-dv/asm_test/riscv_arithmetic_basic_test_0.S', '-I$HOME/repositories/other_repositories/github/riscv-dv/user_extension', '-T$HOME/repositories/other_repositories/github/riscv-dv/scripts/link.ld', '-o', '$HOME/temporary/riscv-dv/asm_test/riscv_arithmetic_basic_test_0.o', '-march=rv32i', '-mabi=ilp32']' returned non-zero exit status 1.
```

I verified the `.S` files generated by `dv` and indeed that is the case. I tried emptying `implemented_csr` in `pygen/pygen_src/target/rv32i/riscv_core_setting.py` but to no avail. Do I need to specify `rv32iZicsr` to `--target`?

5hayanB, 2023-11-24:
I looked further into the problem and have found out that the `riscv-dv` is generating the wrong `.S` file. On compiling the the assembly file with

```zsh
riscv64-unknown-elf-gcc -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles riscv_arithmetic_basic_test_0.S -I/dv/user_extension -T/dv/scripts/link.ld -o riscv_arithmetic_basic_test_0.o -march=rv32izicsr -mabi=ilp32
```

the compilation works and `.o` object file is being created successfully. Why would `riscv-dv` generate a the wrong assembly test when the `--target rv32i` is already given? The `riscv_arithmetic_basic_test` field of the testlist yaml even has `+no_csr_instr=1`, meaning no CSR instruction is to be created.

5hayanB, 2023-11-24:
Seems like `riscv-dv` broke with `riscv64-unknown-elf-gcc` version `13.2.0` as I tested `riscv-dv` with `riscv64-unknown-elf-gcc` version `11.1.0` and is working. I am now changing issue topic to the specific problem at hand.

5hayanB, 2023-11-26:
After loads of compiling and testing different tags of the `[riscv-gnu-toolchain](https://github.com/riscv-collab/riscv-gnu-toolchain/tree/master)`, tags `2023.04.14` and up are incompatible with `riscv-dv` as of yet. I do not know if I should leave the issue open to keep track of toolchain compatibility or close it to after finding out the working version. I will leave the issue open and let the maintainers take the required action as they see fit.

Sai-Manish, 2024-05-20:
Hi @5hayanB, Could you figure out which version is working or why it was not working?

5hayanB, 2024-07-06:
@Sai-Manish, I was able to find that the tags `2023.03.14` and below of the [riscv-gnu-toolchain](https://github.com/riscv-collab/riscv-gnu-toolchain) work with `riscv-dv`. I think the problem is that `riscv-dv` uses initial csr instructions in program initialization and assumes that the `rv32i` extension supports csr extensions. However, the spec now states that the csr extensions be implemented in `Zicsr` which is not a part of `rv32i` which is also updated in the latest riscv-gnu-toolchain.