chipsalliance/riscv-dv #967: Error "SolveBeforeMustBeRand" while Elaborating with Dsim tool

magnetworks, 2024-02-05:
Hi,
I'm trying to build & run the repo using DSIM. 
I'm get the following error - 

```
=E:[SolveBeforeMustBeRand]:
    The variables mentioned in a solve-before constraint
    must be simple variables (no indexes or members)
    and must be rand (not state or randc).

    ./src/riscv_pmp_cfg.sv:106:7
    Included from src/riscv_instr_pkg.sv:1529:12
```

Any help with this would be great.

pdhakal-metrics, 2024-05-04:
Here's the problematic line:
`solve mseccfg.mml before pmp_cfg[i].w, pmp_cfg[i].r;`

`mseccfg` is a rand variable of a packed struct. `dsim` does not currently allow slices/members of packed structs in the "solve before" directive. It would be a substantive rewrite of the code, which is not in the immediate pipeline. Would solving for `mseccfg` instead of a specific bit within it be an acceptable workaround?
`solve mseccfg before pmp_cfg[i].w, pmp_cfg[i].r;`

mcandress, 2024-06-04:
There should be an upcoming enhancement to DSim that will allow more slices/members to be used in solve/before constraints and will solve this issue. 

Meanwhile, you can get the desired behaviour by adding a variable that is equal to mseccfg.mml and using it in the solve-before.
```

rand bit tmpVar;
...
constraint xwr_c {
    tmpVar == msccfg.mml; 
    foreach (pmp_cfg[i]) {
        solve tmpVar before pmp_cfg[i].w, pmp_cfg[i].r
      }
     ...
  }
```