chipsalliance/riscv-dv #968: Error: unrecognized opcode `csrr x5,0xf14', extension `zicsr' required

AhmedAmrAbdellatif1, 2024-02-06:
Hello everyone,
I was trying to run a single test `run --test=riscv_arithmetic_basic_test --simulator questa`
but these bunch of errors keep showing up
Configuration I am using,
OS – Ubuntu22.04
Python – 3.10
gcc – 13.1
[image]
Please I seek Immediate guidance on this.

I want to mention that i need to use RISCV-DV to design my RV64I architecture without CSR being implemented and i don't know how to run tests on my design files

[test.log](https://github.com/chipsalliance/riscv-dv/files/14183483/test.log)

MikeOpenHWGroup, 2024-02-06:
Hi @AhmedAmrAbdellatif1, these are toolchain, not riscv-dv errors.  riscv-dv is generating instructions that require the Zicsr extenion on your invocation of your toolchain (most likely gcc).  You will need to add "zicsr" to -march string.

AhmedAmrAbdellatif1, 2024-02-06:
Hi @MikeOpenHWGroup, thank you for the fast response. So, you're suggesting that there's a problem with the toolchain. which cause errors during testing, right?
If so, what do you suggest?

>You will need to add "zicsr" to -march string.
And sorry i didn't understand which do you mean. could you further explain?

Thanks in advance.

MikeOpenHWGroup, 2024-02-06:
> sorry i didn't understand which do you mean. could you further explain?

Sure.  I should point out that I am a _user_ of riscv-dv, not a _developer_ and I do not use the `run.py`.  So with that caveat...

Your invocation of `riscv-dv` is generating a test-program with CSR access instructions in it (e.g.`csrr x5, 0xf14`).  I am not a toolchain expert, but I believe that the CSR access instructions are _not_ part of the default "march" of older version of gcc for RISC-V.  So you may need to update your toolchain or pass an updated "march" argument to compile this test-program. What version of gcc are you running?

If you cannot upgrade your gcc you should be able to invoke the proper march from the run.py command-line.  Again, I do not use run.py, so to this point I am less certain: there is command-line argument to `run.py` that allows you to set the ISA.  Try: `run.py --isa rv32imc_zicsr`.

AhmedAmrAbdellatif1, 2024-02-06:
I see. But what if i want to exclude the generation CSR Instructions from RISCV DV? currently i want to test my design which's RV64I without CSR Instructions. so even if i solve the problem of toolchain, i would have another program which's i want to eliminate CSR instructions from the base I test.

> What version of gcc are you running?
>
I'm using [https://github.com/riscv-collab/riscv-gnu-toolchain](RISC-V GNU Compiler Toolchain) v13.2

Thanks @MikeOpenHWGroup  in advance.

AhmedAmrAbdellatif1, 2024-02-07:
Hello @zeeshanrafique23, I did so but there're some errors:
```
juba@juba-virtual-machine:~/riscv-dv$ run --test=riscv_arithmetic_basic_test -cs target/rv64i --simulator questa 
Wed, 07 Feb 2024 08:02:55 INFO     Creating output directory: out_2024-02-07
Wed, 07 Feb 2024 08:02:55 INFO     Processing regression test list : target/rv64i/testlist.yaml, test: riscv_arithmetic_basic_test
Wed, 07 Feb 2024 08:02:55 INFO     Processing regression test list : /home/juba/riscv-dv/yaml/base_testlist.yaml, test: riscv_arithmetic_basic_test
Wed, 07 Feb 2024 08:02:55 INFO     Found matched tests: riscv_arithmetic_basic_test, iterations:2
Wed, 07 Feb 2024 08:02:55 INFO     Processing simulator setup file : /home/juba/riscv-dv/yaml/simulator.yaml
Wed, 07 Feb 2024 08:02:55 INFO     Found matching simulator: questa
Wed, 07 Feb 2024 08:02:55 INFO     Building RISC-V instruction generator
Wed, 07 Feb 2024 08:03:12 INFO     Running RISC-V instruction generator
Wed, 07 Feb 2024 08:03:12 INFO     Generating 2 riscv_arithmetic_basic_test
Wed, 07 Feb 2024 08:03:12 INFO     Running riscv_arithmetic_basic_test with 1 batches
Wed, 07 Feb 2024 08:03:12 INFO     Running riscv_arithmetic_basic_test, batch 1/1, test_cnt:2
Wed, 07 Feb 2024 08:03:27 INFO     Compiling out_2024-02-07/asm_test/riscv_arithmetic_basic_test_0.S
riscv64-unknown-elf-gcc: error: missing argument to '-march='
riscv64-unknown-elf-gcc: error: missing argument to '-mabi='
Traceback (most recent call last):
  File "/home/juba/.local/bin/run", line 33, in <module>
    sys.exit(load_entry_point('riscv-dv', 'console_scripts', 'run')())
  File "/home/juba/riscv-dv/run.py", line 1149, in main
    gcc_compile(matched_list, output_dir, args.isa, args.mabi,
  File "/home/juba/riscv-dv/run.py", line 457, in gcc_compile
    run_cmd_output(cmd.split(), debug_cmd=debug_cmd)
  File "/home/juba/riscv-dv/scripts/lib.py", line 207, in run_cmd_output
    raise exc
  File "/home/juba/riscv-dv/scripts/lib.py", line 204, in run_cmd_output
    output = subprocess.check_output(cmd)
  File "/usr/lib/python3.10/subprocess.py", line 421, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/home/juba/riscv-demo//bin/riscv64-unknown-elf-gcc', '-static', '-mcmodel=medany', '-fvisibility=hidden', '-nostdlib', '-nostartfiles', 'out_2024-02-07/asm_test/riscv_arithmetic_basic_test_0.S', '-I/home/juba/riscv-dv/user_extension', '-T/home/juba/riscv-dv/scripts/link.ld', '-o', 'out_2024-02-07/asm_test/riscv_arithmetic_basic_test_0.o', '-march=', '-mabi=']' returned non-zero exit status 1.
```

AhmedAmrAbdellatif1, 2024-02-07:
Hello everyone,
I've solved the problem. 
As @MikeOpenHWGroup mentioned, the problem is related to the [RISC-V GNU Compiler Toolchain](https://github.com/riscv-collab/riscv-gnu-toolchain). As the latest update has excluded the CSR-related instructions from the RVGC to external extensions: `zicsr` and `zifence`
So to solve the problem you two ways while building the toolchain

1. to include the `zicsr` and `zifencei` extensions. for example:
```
./configure --prefix=/opt/riscv --with-arch=rv64gc_zicsr_zifencei
make
```

2. to use the older ISA spec version, however this is strongly *NOT* recommended.
```
./configure --prefix=/opt/riscv --with-isa-spec=2.2
make
```

I tried the latter option and it work for me.

Thanks -

closed 2024-02-07 by AhmedAmrAbdellatif1

MikeOpenHWGroup, 2024-02-07:
Glad to hear you are up and running @AhmedAmrAbdellatif1.  Perhaps you would consider creating a pull-request to update the documentation?  (Note: I am not a member of Chips Alliance, so it will fall to others to review/approve/merge such a PR.)

MubashirSaleem775, 2024-08-28:
@AhmedAmrAbdellatif1  I have tried both ways but still not able to resolve the issue.

1519144909, 2024-09-11:
> 大家好，我已经解决了问题。如前所述，该问题与@MikeOpenHWGroup[RISC-V GNU 编译器工具链](https://github.com/riscv-collab/riscv-gnu-toolchain).由于最新的更新已将 RVGC 的 CSR 相关指令排除到外部扩展：因此，在构建工具链时，您可以通过两种方式来解决问题`zicsr``zifence`
> 
> 1. 以包括 和 扩展。例如：`zicsr``zifencei`
> 
> ```
> ./configure --prefix=/opt/riscv --with-arch=rv64gc_zicsr_zifencei
> make
> ```
> 
> 但是，我试过了，但没有用
> 
> 2. 使用较旧的 ISA 规范版本，但是强烈建议_不要_这样做。
> 
> ```
> ./configure --prefix=/opt/riscv --with-isa-spec=2.2
> make
> ```
> 
> 我尝试了后一种选择，它对我有用。
> 
> 谢谢-
configure 在哪里？