chipsalliance/riscv-dv #970: How to connect the test to my SV design file, and How do I get the feedback that the test has passed or failed?

AhmedAmrAbdellatif1, 2024-02-07:
Hello everyone, 

I've designed an RISCV architecture and i want to test it using `RISCV-DV`.
So, is there anyone can guide me what should i exactly do to test my .sv design files and how to get the feedback of the passing or failing?

Thanks in advance.

MikeOpenHWGroup, 2024-02-07:
Hi @AhmedAmrAbdellatif1.  I think what you are asking about is outside the scope of riscv-dv.  There are several open-source implementations of verification environments that you could look at:
- [This](https://riscv.org/wp-content/uploads/2018/12/14.25-Tao-Liu-Richard-Ho-UVM-based-RISC-V-Processor-Verification-Platform.pdf) presentation by the original developers of riscv-dv provides an overview of how riscv-dv can be used to create a self-checking testbench environment.  (I do not know if an example implementation of that testbench is available.)
- The [Ibex](https://github.com/lowRISC/ibex) core, managed by lowRISC has an excellent SV/UVM environment.
- [CORE-V-VERIF](https://github.com/openhwgroup/core-v-verif) is the OpenHW Group's SV/UVM environment for multiple cores (including one that is a fork of Ibex).
- [Wally](https://github.com/openhwgroup/cvw) is another OpenHW project that uses a different SV environment.

And of course there are many commercial products available as well (some of which integrate riscv-dv!).

The goals of each of these are slightly different from one another, so you'll want to consider your needs and the goals of your project before deciding on a path forward.

LF-RoGu, 2025-03-25:
Hi @AhmedAmrAbdellatif1 another point of view if it helps, the point of RISCV-DV is simply to be a random instruction generator, it already provides some testbench frameworks to work with. 

What it simply does is generating a wide variety of test cases and scenarios to stress different parts of the processor pipeline. (at least to my understanding, as I'm also trying to learn how to work with this tool)
And whatever comes out, it will compare it with an ISS (Golden model) like spike for you to validate if your system does not present bugs.
[In really simple words, uses various test scenarios like the random instruction generator to try and break your design, however your design should already be working and at least tested by you, this is mainly to catch bugs and verify that your design behaves as intended]

It will not tell you if your design is working or not, it simply stresses your system for you to check what could be improved.

PS: If someone sees this thread please correct me if I'm wrong so this does not create a misleading answer for someone in the future.

Also the previous link refered by @MikeOpenHWGroup does not have the presentation or is no longer available. I found the same version here: https://www.slideshare.net/slideshow/uvmbased-riscv-processor-verification-paltform-pdf/252362017