chipsalliance/riscv-dv #975: Can't do S-mode test ( No delegation )

AhmedAmrAbdellatif1, 2024-04-13:
Hello everyone,
I've design an RV64IMAC architecture with M & S Privilege levels supported.
I want to test the S-mode, I see that the assembly generated has S-mode exception handler and stvec handler, but spike only does M-mode handler exception and mtvec handler.

How can i force spike to go to s-mode exception handler?

Also, how i force spike to use `sscratch` instead of `mscratch`?

MikeOpenHWGroup, 2024-04-13:
Hi @AhmedAmrAbdellatif1, I am not a member of ChipsAlliance, but I do not believe they maintain/support Spike.  I suggest you refer to the [Spike documentation](https://github.com/riscv-software-src/riscv-isa-sim/blob/master/README.md).  You could also check out this [Spike documentaton effort](https://github.com/poweihuang17/Documentation_Spike).

AhmedAmrAbdellatif1, 2024-04-13:
Hello @MikeOpenHWGroup, Maybe i wasn't clear enough. but my question is "how to generate test that uses stvec handler and s-mode exception and stuff"

serge0699, 2024-08-25:
Hello @AhmedAmrAbdellatif1, you can use `init_privileged_mode` option. See [`riscv_instr_gen_config.sv`](https://github.com/chipsalliance/riscv-dv/blob/master/src/riscv_instr_gen_config.sv) and [`riscv_asm_program_gen.sv`](https://github.com/chipsalliance/riscv-dv/blob/master/src/riscv_asm_program_gen.sv).