chipsalliance/riscv-dv #976: Randomize CSR in main

predator-111, 2024-04-24:
riscv-dv initializes floating point CSRs frm using init_floating_point_gpr(). I want to add randomization to certain CSRs in the main function without breaking the existing setup. Kindly suggest any way to achieve this effectively.

zeeshanrafique23, 2024-04-24:
Not sure, if this is helpful to you:
https://htmlpreview.github.io/?https://github.com/google/riscv-dv/blob/master/docs/build/singlehtml/index.html#setup-privileged-csr-description-optional

MikeOpenHWGroup, 2024-04-24:
You can extend `riscv_asm_program_gen` and overload the `init_floating_point_gpr()` method:
```
class my_riscv_asm_program_gen extends riscv_asm_program_gen;

  // my implementation of init_floating_point_gpr()
  virtual function void init_floating_point_gpr();
    int int_gpr;
    string str;
    for(int i = 0; i < NUM_FLOAT_GPR; i++) begin
      // your code goes here...
    end
    // Initialize rounding mode of FCSR
    str = $sformatf("%0sfsrmi %0d", indent, cfg.fcsr_rm);
    instr_stream.push_back(str);
  endfunction

endclass: my_riscv_asm_program_gen
```
You will also need to create your own build script(s) to add the new class to the manifest.

Warning!  I have based the above on what we have done in the past, and the above code has not been exposed to a SystemVerilog compiler.  :stuck_out_tongue_closed_eyes:

predator-111, 2024-04-25:
This would add the instructions that I want in the initialization section of the program. What I want to achieve is the CSR instructions appearing randomly in the main section of the program, not just in the initialization section

MikeOpenHWGroup, 2024-04-25:
You can apply the same strategy to any class (and thereby class function/task) in riscv-dv.

predator-111, 2024-04-30:
When I am trying to extend a particular class and trying to accommodate fsrmi instruction (or any particular CSR instruction), the types of queues there are of the riscv_instr_t, and this riscv_instr_t does not support CSR instructions, as those instructions can not be casted into that type (that is why those CSR instructions are hardcoded as a string and used in riscv_asm_program_gen.sv, the queue type being there is string, hence any string can be pushed, but only in the initialization section of the program). Hence, I can not push those CSR instructions in the main part of the program by extending any class.

MikeOpenHWGroup, 2024-04-30:
> Hence, I can not push those CSR instructions in the main part of the program by extending any class.

Ah, now I get it.  I'll bet ChipsAlliance would look favourably upon a pull-request to update `riscv_instr_t`.  (Disclaimer, I am not a member of CA.)