chipsalliance/riscv-dv #979: Can riscv-dv used with ucb rocket core/boom core?

Sai-Manish, 2024-05-15:
I would like to understand more on how to integrate other riscv-dv for other cores.
Can I find much more detailed documentation on how to use it?

davine47, 2024-05-15:
Hi, riscv-dv is a static random instruction generator, which means that final outputs are bin files.
Rocket-chip core provides a instruction bootrom in [ExampleRocketSystem.scala](https://github.com/chipsalliance/rocket-chip/blob/master/src/main/scala/system/ExampleRocketSystem.scala), it will read the bin file and then the core can fetch the target instructions.

Sai-Manish, 2024-05-16:
Hi @davine47, Thanks for your reply.
I have few more questions,
1) Can we use verilator for the yaml flow? If yes, is there any compiler.yaml file which has verilator setup in it?
2) Instead of adding the bin file into bootrom is there any other way I can verify the core? like similar to torture tests in`[as in the repo](https://github.com/ucb-bar/riscv-torture/tree/b2b66a66d51b360e0ae95017774d03377c78c574)`.

franAyachi, 2024-09-23:
> Hi @davine47, Thanks for your reply. I have few more questions,
> 
>     1. Can we use verilator for the yaml flow? If yes, is there any compiler.yaml file which has verilator setup in it?
> 
>     2. Instead of adding the bin file into bootrom is there any other way I can verify the core? like similar to [torture tests.
>        ](https://github.com/ucb-bar/riscv-torture/tree/b2b66a66d51b360e0ae95017774d03377c78c574)
> 
>     3. I am not able generate the binary file following the steps in README.md

Hi!, Have you been able to use verilator as you asked? Thanks!

MikeOpenHWGroup, 2024-09-23:
Hi @franAyachi, although Verilator does not yet support the UVM (as of v5.029, anyway), progress is being made.  If you like, you can use [this repo](https://github.com/MikeCovrado/GettingVerilatorStartedWithUVM) to test it out for yourself.  If you do not have a commerical SystemVerilog simulator capable of handling UVM, you could try [Metrics DSim](https://www.metrics.ca/), a free-to-use commercial grade simulator.  At one point the riscv-dv scriptware supported DSim, but there may have been some bit-rot.  If that is the case, please open and issue here and I can help.  (Note, neither I or the OpenHW Group are members of the ChipsAlliance, but we are a big users of riscv-dv.)

franAyachi, 2024-10-08:
> Hi @franAyachi, although Verilator does not yet support the UVM (as of v5.029, anyway), progress is being made. If you like, you can use [this [...]

Hi @MikeOpenHWGroup!  Im using the UVM library adapted to try riscv-dv with Verilator as you said. Also adapted the yaml file like this: 
`- tool: verilator
  compile:
    cmd:
      - verilator
              -DUVM_NO_DPI
              -DUVM_REGEX_NO_DPI
              -DUVM_REPORT_DISABLE_FILE_LINE
              -DSVA_ON
              --binary 
              --cc
              --exe
              --hierarchical
              -CFLAGS '-std=c++11'
              -LDFLAGS '-Wl,--no-as-needed'
              -o <out>/simv <cmp_opts>
              -Wno-DECLFILENAME
              -Wno-CONSTRAINTIGN 
              -Wno-MISINDENT 
              -Wno-VARHIDDEN 
              -Wno-WIDTHTRUNC 
              -Wno-CASTCONST 
              -Wno-WIDTHEXPAND 
              -Wno-UNDRIVEN 
              -Wno-UNUSEDSIGNAL 
              -Wno-UNUSEDPARAM 
              -Wno-ZERODLY 
              -Wno-SYMRSVDWORD 
              -Wno-CASEINCOMPLETE 
              -Wno-REALCVT
              -I<setting>
              -I<user_extension>
              -I/src
              -I/test
              -Iuvm-verilator/src/ 
              uvm-verilator/src/uvm_pkg.sv 
              -f <cwd>/files.f > compile_transcript_Verilator.txt 2>&1`

After many changes Im block with this same issue: https://github.com/verilator/verilator/issues/4497 
But the thing its that riscv-dv doesnt use any sequencer, so I don´t know if there is any way to solve this issue for riscv-dv.

Thanks!

MikeOpenHWGroup, 2024-10-08:
@franAyachi, my advice is to not attempt to use Verilator with riscv-dv.  They are working on it, but it will be a while before Verilator is able to compile and simulate all of the IEEE 1800-207 constructs required to support UVM.  Have you tried DSim?