chipsalliance/riscv-dv #997: Generate load store instruction for particular range

ankurcere, 2025-01-09:
I am looking for a way to generate load and store instructions specifically targeting a particular range of memory addresses using RISC-V DV. Is this currently supported, or would it be possible to implement?

MikeOpenHWGroup, 2025-01-09:
A good place to start is the [Configuration](https://htmlpreview.github.io/?https://github.com/google/riscv-dv/blob/master/docs/build/singlehtml/index.html#document-configuration) section of the user documentation.  Take a look at the `Setup the memory map` sub-section.

From there you'll need to read the source code. 😉 Start with the simple class extension of [riscv_load_store_instr_lib.sv](https://github.com/chipsalliance/riscv-dv/blob/f0c570d11236f94f9c5449870223a5ac717cc580/src/riscv_load_store_instr_lib.sv#L213).  That file has a number of interesting extensions you can use to generate your test-program.  A particularly interesting one is [riscv_load_store_rand_addr_instr_stream](https://github.com/chipsalliance/riscv-dv/blob/f0c570d11236f94f9c5449870223a5ac717cc580/src/riscv_load_store_instr_lib.sv#L428).

ankurcere, 2025-01-13:
Thanks,

But I have to do this for different harts supposed to read hart id = 1 by using some constraint, then define load and store access from memory region 0x0 to 0x1000 and for hart id == 2 the address space will be 0x1000 to 0x2000 and so on, to completely ignore the race condition and generate load store to specific location specific to each hart. Can I do this?

ankurcere, 2025-01-22:
So, how riscv-dv manage race condition in the case of multi-harts? Is it supported multi-harts fully?

ankurcere, 2025-01-23:
Thanks. I will raise another issue.

closed 2025-01-23 by ankurcere